pyocd/cmsis-pack-manager #86: After Win10 clean mbed CLI install (Mbed_installer_v0.4.10.exe), CMSIS errors

j3hill, 2019-03-25:
```
D:\Dev\mbed-os-example-ble\BLE_HeartRate>mbed compile -t ARM -m NRF52840_DK
[mbed] Working path "D:\Dev\mbed-os-example-ble\BLE_HeartRate" (library)
[mbed] Program path "D:\Dev\mbed-os-example-ble\BLE_HeartRate"
[mbed] Auto-installing missing Python modules (cmsis_pack_manager)...
Traceback (most recent call last):
  File "D:\Dev\mbed-os-example-ble\BLE_HeartRate\mbed-os\tools\make.py", line 41, in <module>
    from tools.options import get_default_options_parser
  File "D:\Dev\mbed-os-example-ble\BLE_HeartRate\mbed-os\tools\options.py", line 24, in <module>
    from .toolchains import TOOLCHAINS, EXTRA_TOOLCHAIN_NAMES
  File "D:\Dev\mbed-os-example-ble\BLE_HeartRate\mbed-os\tools\toolchains\__init__.py", line 19, in <module>
    from . import mbed_toolchain, arm, gcc, iar
  File "D:\Dev\mbed-os-example-ble\BLE_HeartRate\mbed-os\tools\toolchains\mbed_toolchain.py", line 48, in <module>
    from ..config import (ConfigException, RAM_ALL_MEMORIES, ROM_ALL_MEMORIES)
  File "D:\Dev\mbed-os-example-ble\BLE_HeartRate\mbed-os\tools\config\__init__.py", line 33, in <module>
    from ..arm_pack_manager import Cache
  File "D:\Dev\mbed-os-example-ble\BLE_HeartRate\mbed-os\tools\arm_pack_manager\__init__.py", line 24, in <module>
    from cmsis_pack_manager import Cache as _Cache
  File "C:\Python27\lib\site-packages\cmsis_pack_manager\__init__.py", line 21, in <module>
    from ._native import ffi, lib
  File "C:\Python27\lib\site-packages\cmsis_pack_manager\_native.py", line 7, in <module>
    lib = ffi.dlopen(os.path.join(os.path.dirname(__file__), '_native__lib.cp37-win32.pyd'), 0)
OSError: cannot load library 'C:\Python27\lib\site-packages\cmsis_pack_manager\_native__lib.cp37-win32.pyd': error 0x7e
[mbed] ERROR: "C:\Python27\python.exe" returned error.
       Code: 1
       Path: "D:\Dev\mbed-os-example-ble\BLE_HeartRate"
       Command: "C:\Python27\python.exe -u D:\Dev\mbed-os-example-ble\BLE_HeartRate\mbed-os\tools\make.py -t ARM -m NRF52840_DK --source . --build .\BUILD\NRF52840_DK\ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
```

theotherjimmy, 2019-03-26:
Huh... it seems to be there...
```
Archive:  cmsis_pack_manager-0.2.4-py2.py3-none-win32.whl
  Length      Date    Time    Name
---------  ---------- -----   ----
     9864  2019-03-15 15:46   cmsis_pack_manager/__init__.py
      206  2019-03-15 16:12   cmsis_pack_manager/_native.py
     1483  2019-03-15 16:12   cmsis_pack_manager/_native__ffi.py
  3095040  2019-03-15 16:12   cmsis_pack_manager/_native__lib.cp37-win32.pyd
     7847  2019-03-15 15:46   cmsis_pack_manager/pack_manager.py
      134  2019-03-15 16:12   cmsis_pack_manager-0.2.4.dist-info/LICENSE
      275  2019-03-15 16:12   cmsis_pack_manager-0.2.4.dist-info/METADATA
      121  2019-03-15 16:12   cmsis_pack_manager-0.2.4.dist-info/WHEEL
       71  2019-03-15 16:12   cmsis_pack_manager-0.2.4.dist-info/entry_points.txt
       19  2019-03-15 16:12   cmsis_pack_manager-0.2.4.dist-info/top_level.txt
     1006  2019-03-15 16:12   cmsis_pack_manager-0.2.4.dist-info/RECORD
---------                     -------
  3116066                     11 files
```

Perhaps this thinks it's 32 bit python, but it's really 64bit python...

thendrickson, 2019-05-20:
I'm still seeing this issue. Any workarounds?

kjsGraspor, 2022-09-13:
@thendrickson 
Hi Travis. Have you by any change solve the issue? I have the exact same problem.. 

Thanks

thendrickson, 2022-09-13:
@kjsGraspor 
It's working for me on windows 10 64-bit and python 3.7.9 64-bit