pyocd/cmsis-pack-manager #87: Some targets still not having flash algorithms being marked as "default"

bridadan, 2019-04-01:
This is specific to CPM's use in Mbed OS. This PR solved a lot of these issues: https://github.com/ARMmbed/cmsis-pack-manager/pull/79

However a few targets still don't have any flash algorithms marked as `default`.

This target (`MK11DN512VMC5`) has its algorithm marked as default:
```xml
<devices>
    <family Dfamily="MK11D5" Dvendor="NXP:11">
      <description>The Kinetis(r) K11 50 MHz baseline MCUs, built on the ARM(r) Cortex(r)-M4 core, are optimized for cost-sensitive applications requiring low-power, security encryption, tamper detection and processing efficiency. This family shares the comprehensive enablement and scalability of the Kinetis portfolio.</description>
      <device Dname="MK11DN512xxx5">
        <processor Dcore="Cortex-M4" Dfpu="0" Dendian="Little-endian" Dclock="50000000"/>
        <environment name="iar">
          <file category="linkerfile" name="iar/MK11DN512xxx5_flash.icf"/>
        </environment>
        <memory name="FLEX_RAM" access="rw" start="0x14000000" size="0x1000"/>
        <memory name="PROGRAM_FLASH" access="rx" start="0x00000000" size="0x080000" startup="1" default="1"/>
        <memory name="SRAM_LOWER" access="rw" start="0x1fff8000" size="0x8000" default="1"/>
        <memory name="SRAM_UPPER" access="rw" start="0x20000000" size="0x8000"/>
        <algorithm name="arm/MK_P512_50MHZ.FLM" start="0x00000000" size="0x00080000" default="1"/>
        <debug svd="MK11D5.xml"/>
        <variant Dvariant="MK11DN512VLK5">
          <compile header="fsl_device_registers.h" define="CPU_MK11DN512VLK5"/>
          <environment name="uv">
            <CMisc>--C99</CMisc>
            <LMisc>--scatter "./RTE/Device/MK11DN512VLK5/MK11DN512xxx5_flash.scf"</LMisc>
          </environment>
        </variant>
        <variant Dvariant="MK11DN512VMC5">
          <compile header="fsl_device_registers.h" define="CPU_MK11DN512VMC5"/>
          <environment name="uv">
            <CMisc>--C99</CMisc>
            <LMisc>--scatter "./RTE/Device/MK11DN512VMC5/MK11DN512xxx5_flash.scf"</LMisc>
          </environment>
        </variant>
      </device>
    <!-- SNIP -->
    </family>
</devices
```

But this target does not (`KW24D`):
```xml
<devices>
  <family Dfamily="KWxx Series" Dvendor="NXP:11">
    <description>
      <!-- SNIP -->
    </description>
    <!-- ******************************  KW2xD  ****************************** -->
    <subFamily DsubFamily="KW2xD">
      <processor Dcore="Cortex-M4" DcoreVersion="r0p0"/>
      <book name="Documents/dui0553a_cortex_m4_dgug.pdf" title="Cortex-M4 Generic User Guide"/>
      <description><!-- SNIP --></description>
      <!-- ******************************  MKW24D512xxx5  ****************************** -->
      <device Dname="MKW24D512xxx5">
        <processor Dfpu="0" Dmpu="0" Dendian="Little-endian" Dclock="50000000"/>
        <compile header="Device/Include/MKW24D5.h" define="MKW24D512xxx5"/>
        <debug svd="SVD/MKW24D5.svd"/>
        <memory id="IROM1"                        start="0x00000000" size="0x00080000" default="1" startup="1"/>
        <memory id="IRAM1"                        start="0x1FFF8000" size="0x00010000" default="1" init   ="0"/>
        <algorithm name="Flash/MK_P512_50MHZ.FLM" start="0x00000000" size="0x00080000" default="1" RAMstart="0x20000000" RAMsize="0x1000"/>

        <book name="Documents/MKW2xDxxx.pdf"   title="MKW24D512xxx5 Data Sheet"/>
        <book name="Documents/MKW2xDxxxRM.pdf" title="MKW24D512xxx5 Reference Manual"/>
      </device>
    </subFamily>
  </family>
</devices>
```

Any idea why the algo for the `KW24D` isn't being picked up?

theotherjimmy, 2019-04-02:
It does have default set on that algorithm:
```
$pack-manager find-part -l MKW24D512xxx5
MKW24D512xxx5:
  algorithms:
  - {default: true, file_name: Flash/MK_P512_50MHZ.FLM, ram_size: 4096, ram_start: 536870912,
    size: 524288, start: 0}
  family: KWxx Series
  from_pack: {pack: Kinetis_KWxx_DFP, url: 'http://www.keil.com/pack/', vendor: Keil,
    version: 1.9.0}
  memories:
    IRAM1:
      access: {execute: false, non_secure: false, non_secure_callable: false, peripheral: false,
        read: true, secure: false, write: true}
      default: true
      size: 65536
      start: 536838144
      startup: false
    IROM1:
      access: {execute: true, non_secure: false, non_secure_callable: false, peripheral: false,
        read: true, secure: false, write: false}
      default: true
      size: 524288
      start: 0
      startup: true
  name: MKW24D512xxx5
  processor:
    Symmetric: {core: CortexM4, fpu: None, mpu: NotPresent, units: 1}
  sub_family: KW2xD
  vendor: NXP:11
```

bridadan, 2019-04-02:
Hmm, it didn't make it into the Mbed OS index.

theotherjimmy, 2019-04-02:
Sounds like an issue for Mbed OS then.

bridadan, 2019-04-15:
Like @theotherjimmy said, this is an Mbed OS issue.

closed 2019-04-15 by bridadan