pyocd/cmsis-pack-manager #91: add sources to pypi

shlomow, 2019-04-13:
Hi!

Can you please upload the source ti pypi server in addition to the wheel files? It can be useful for some use cases. Thanks in advance!

Btw, if I am trying to install the source from the releases the tab I am getting en error when building the package. I am using win10.

flit, 2019-04-13:
What error do you get when building?

shlomow, 2019-04-14:
@flit This is the output:

```
Installing collected packages: milksnake, cmsis-pack-manager
  Running setup.py install for cmsis-pack-manager ... error
    Complete output from command c:\users\shlomi\appdata\local\programs\python\python37\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\shlomi\\AppData\\Local\\Temp\\pip-req-build-qa6pp_wp\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\shlomi\AppData\Local\Temp\pip-record-mbx2ps5x\install-record.txt --single-version-externally-managed --compile:
    fatal: not a git repository (or any of the parent directories): .git
    running install
    running build
    running build_py
    creating build\lib
    creating build\lib\cmsis_pack_manager
    copying cmsis_pack_manager\pack_manager.py -> build\lib\cmsis_pack_manager
    copying cmsis_pack_manager\__init__.py -> build\lib\cmsis_pack_manager
    error: [WinError 2] The system cannot find the file specified

    ----------------------------------------
Command "c:\users\shlomi\appdata\local\programs\python\python37\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\shlomi\\AppData\\Local\\Temp\\pip-req-build-qa6pp_wp\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\shlomi\AppData\Local\Temp\pip-record-mbx2ps5x\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in C:\Users\shlomi\AppData\Local\Temp\pip-req-build-qa6pp_wp\
```

flit, 2019-04-18:
I'm not sure what that error could be caused by since it doesn't say what file could not be found. Do you have Rust installed? It could be complaining about not finding the `cargo` tool.

cederom, 2019-04-28:
Hello @flit :-) I was about to write the same issue / question - could you please provide source package to pip/pypi ? At the moment only binary packages are provided. This package is a mandatory requirement of mbed-os. On FreeBSD this package is unavailable, so using mbed-os is not possible. Having source package at pypi would make it possible to compile the dependency during pip install..

theotherjimmy, 2019-05-07:
@shlomow The reason that I don't upload source packages is because they are not immediately usable on every platform without an installation of rustc + cargo. Uploading the source could cause pip to try to download and run cargo (giving you the error you described) on these systems.

cederom, 2019-05-07:
Hey there @theotherjimmy :-) Binary only distribution limits usage of CMSIS Pack Manager to provided platforms - for instance I cannot use it at all on FreeBSD..

Can we work out the build automation please? I am eager to help :-)

theotherjimmy, 2019-05-31:
@cederom I forgot to respond here. Yes, I'm the author; Yes we can work out build automation for more platforms. I don't know of a service that would be able to provide me with a way to build binary wheels for freebsd.

cederom, 2019-05-31:
Hey @theotherjimmy, this is some old thread, the work is already done in https://github.com/ARMmbed/cmsis-pack-manager/issues/97, build well, just need to add source package to pypi (https://github.com/ARMmbed/cmsis-pack-manager/issues/112) and thing should work with some local os dependencies for build (such as Rust) :-)

theotherjimmy, 2019-07-31:
I actually din this yesterday, but the source packages contain a bug. I'll  close this issue when I can get a working 0.2.x release of CPM with source that installs as the same release version.