pyocd/cmsis-pack-manager #92: CPM crashing on Windows

bridadan, 2019-04-15:
Running version 0.2.6 in Python 2.7.13 on Windows 10.

The python instance crashes for me (Windows has a pop up showing that `python.exe` has stopped working).

The log for me looks like this:
```
Apr 15 15:41:20.997 WARN No Core found!, Package: CMSIS, Vendor: ARM
Apr 15 15:41:21.232 ERRO when trying to parse component: child element "description" not found in "component" element, Package: RS14100_DFP, Vendor: Redpine
Apr 15 15:41:21.260 WARN invalid digit found in string, Package: TLE984x_DFP, Vendor: Infineon
memory allocation of 80359 bytes failed
```

bridadan, 2019-05-06:
Some extra info:

On Windows, I can't run `pack-manager cache everything` without it running out of memory:

```
$ pack-manager cache everything
May 06 09:34:41.548 ERRO A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond. (os error 10060), uri: http://www.mindmotion.com.cn/Download/MDK_KEIL/MindMotion.pidx
May 06 09:34:41.906 ERRO A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond. (os error 10060), uri: http://wcnbbs.unisoc.com/data/cmsis-pack/Unisoc.pidx
May 06 09:34:47.615 ERRO when trying to parse component: child element "description" not found in "component" element, Package: RS14100_DFP, Vendor: Redpine
May 06 09:34:47.727 WARN invalid digit found in string, Package: TLE984x_DFP, Vendor: Infineon
May 06 09:34:49.001 WARN No Core found!, Package: CMSIS, Vendor: ARM
May 06 09:34:49.085 WARN expecting </link> found </head>
May 06 09:34:49.090 WARN expecting </link> found </head>
memory allocation of 81107 bytes failed
```

Throughout the runtime of the command, the memory increases by about 4 MB every second or so, up to about 2GB where I get the memory allocation error above.

Running `pack-manager cache descriptors` seems to work fine, no issues with the memory leak.

theotherjimmy, 2019-05-07:
This seems like it may be a windows-only problem. I can't reproduce it on my mac or linux boxes.

bridadan, 2019-05-07:
@theotherjimmy I humbly offer my laptop to you for debugging purposes 😄