pyocd/cmsis-pack-manager #93: `raw_input` not defined when attempting to download pack via tool

flit, 2019-04-18:
Running `pack-manager cache devs STM32F401RETx` results in this exception:

```
Traceback (most recent call last):
  File "/Users/creed/projects/pyOCD-flit/venv/mac37/bin/pack-manager", line 10, in <module>
    sys.exit(main())
  File "/Users/creed/projects/pyOCD-flit/venv/mac37/lib/python3.7/site-packages/cmsis_pack_manager/pack_manager.py", line 235, in main
    args.command(args)
  File "/Users/creed/projects/pyOCD-flit/venv/mac37/lib/python3.7/site-packages/cmsis_pack_manager/pack_manager.py", line 94, in thunk
    return command(cache, **argv)
  File "/Users/creed/projects/pyOCD-flit/venv/mac37/lib/python3.7/site-packages/cmsis_pack_manager/pack_manager.py", line 148, in command_cache
    raw_input("Press Control-C to quit; Press Enter to continue")
NameError: name 'raw_input' is not defined
```

theotherjimmy, 2019-05-07:
Hmmmm... I should do a better job of testing on python3.

flit, 2019-05-15:
I've used `six.moves.input()` as a compatible replacement for `raw_input()`.

theotherjimmy, 2019-05-15:
I'll do that.

PR #163 (2021-04-03): pack-manager: removed query before pack download
Removed the "press enter to continue" before proceeding to download packs when running `pack-manager cache devs`. This removes the use of `raw_input()`.

Fixes #93
commit 8f9995a1 (2021-04-03): pack-manager: removed query before pack download.

This removes the use of raw_input() and thus resolves #93.
closed 2021-04-03 by flit