pyocd/cmsis-pack-manager #95: Failure to download STM32F4xx DFP

flit, 2019-04-18:
If the Python API is used to download the Keil::STM32F4xx_DFP::2.13.0 pack, you get this file instead:

```xml
<?xml version="1.0" encoding="utf-8"?><Error><Code>BlobNotFound</Code><Message>The specified blob does not exist.
RequestId:00ffc5ad-601e-00b2-3a34-f61678000000
Time:2019-04-18T22:14:45.2212973Z</Message></Error>
```

Downloading manually from http://www.keil.com/dd2/pack works fine.

flit, 2019-04-30:
Fyi, created internal issue SDMDK-8487. Not sure if the problem is with cpm or the Keil server.

flit, 2019-05-02:
Root cause identified: If the URL in the <url> element in the PDSC does not have a trailing slash, CPM fails to properly construct the pack's full URL. Confirmed by editing the `Keil.STM32F4xx_DFP.2.13.0.pdsc` file in the CPM index and re-downloading, which was successful.

theotherjimmy, 2019-05-06:
Oh, interesting. I'll have to make sure that the trailing slash is added correctly.

theotherjimmy, 2019-05-07:
Note to self: `pack-manager cache devs STM32F411RETx` reproduces the behavior.

theotherjimmy, 2019-05-07:
https://github.com/ARMmbed/cmsis-pack-manager/blob/17e6a6d6f3f99c50c1211152e0d0df706649887d/rust/cmsis-update/src/dl_pack.rs#L22 *facepalm* Fix incoming.

PR #100 (2019-05-07): Correct pack downloading when the "/" suffix is ommitted from the URL
This was a copy paste mistake...

Resolves #95
closed 2019-05-07 by theotherjimmy