pyocd/cmsis-pack-manager #96: Index keys need to support case-insensitive compares

flit, 2019-04-18:
The keys for `Cache.index` should be all lower-case so that devices can be found with a case-insensitive match.

theotherjimmy, 2019-05-07:
This is a good idea.

theotherjimmy, 2019-05-14:
@jkrech Are `Dname`s and `Dvariants` required to be unique after lowercasing? Could making all devices names lower case lead to conflicts of legitimate devices down the line?

jkrech, 2019-05-14:
CMSIS-Pack is case sensitive and all we are prescribing in the specification is that each Dvendor is responsible for the uniqueness of the Dname / Dvariant  descriptions distributed by their own packs. This shall ensure that the touple
Dvendor:Dname 
is unique. Therefore in theory lower-casing Dname may result in duplicates. I have not seen that yet but as far as I can see we must not rule it out completely.

theotherjimmy, 2019-05-14:
Thanks. That's helpful. @flit We may have to find another way to search case-insensitively. I also need to have `Dvendor` in the index as well.

flit, 2019-05-15:
Ok, well, I've already implemented case-insensitivity in pyOCD, anyway.

More generally, I agree the DFP attributes must be case-sensitive. Those contain device vendor names or part numbers.

But from a usability perspective in pyOCD, making target type names case-insensitive is a must.