pyocd/cmsis-pack-manager #97: build problem on FreeBSD

cederom, 2019-04-28:
Hello world, I have problem building on FreeBSD.. any hints welcome :-)

```
(venv37bsd) python setup.py -v install
fatal: not a git repository (or any parent up to mount point /ztuff)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
running install
running bdist_egg
running egg_info
writing cmsis_pack_manager.egg-info/PKG-INFO
writing dependency_links to cmsis_pack_manager.egg-info/dependency_links.txt
writing entry points to cmsis_pack_manager.egg-info/entry_points.txt
writing requirements to cmsis_pack_manager.egg-info/requires.txt
writing top-level names to cmsis_pack_manager.egg-info/top_level.txt
'license_file' option was not specified
reading manifest file 'cmsis_pack_manager.egg-info/SOURCES.txt'
writing manifest file 'cmsis_pack_manager.egg-info/SOURCES.txt'
installing library code to build/bdist.freebsd-12.0-RELEASE-p3-amd64/egg
running install_lib
running build_py
not copying cmsis_pack_manager/__init__.py (output up-to-date)
not copying cmsis_pack_manager/pack_manager.py (output up-to-date)
error: [Errno 2] No such file or directory: 'cargo': 'cargo'

(venv37bsd) pip show cargo
Name: cargo
Version: 0.1.1
Summary: Transform dictionaries into Python objects
Home-page: github.com/eugene-eeo/cargo
Author: Eugene Eeo
Author-email: packwolf58@gmail.com
License: UNKNOWN
Location: /ztuff/cederom/work/CeDeROM/work/tixon/lab/CeDeROM_SZULPIK/lab/venv37bsd/lib/python3.7/site-packages
Requires:
Required-by:

(venv37bsd) python --version
Python 3.7.3

(venv37bsd) uname -a
FreeBSD 0xCFC2 12.0-RELEASE-p3 FreeBSD 12.0-RELEASE-p3 GENERIC  amd64

(venv37bsd) ls
LICENSE                         appveyor.yml                    cmsis_pack_manager.egg-info     setup.py
README.md                       build                           requirements.txt                setup_requirements.txt
apache-2.0.txt                  build-wheels.sh                 rust                            test_requirements.txt
appveyor.bat                    cmsis_pack_manager              setup.cfg                       tests
(venv37bsd)
```
Would be really nice if cmsis-pack-manager was also provided in PyPI as sources so it can be automtically compiled by pip on platforms that have no binaries provided :-)

cederom, 2019-04-28:
Fortunately `cmsis-pack-manager` does not seem mandatory for `bmed-os` build..? The firmware started compiling with this package/dependency omitted.. seems enough for me at the moment :-)

Still, it would be nice to make pip package build from sources on different platforms :-)

flit, 2019-04-29:
The cpm author is currently on vacation, but I'll do what I can to help… 😄 

The core of cpm is written in Rust. 'cargo' is the Rust package manager and build tool. The `build-wheels.sh` script has the detailed process building everything. The main requirement is that you install the specific nightly version of Rust listed in the script via `rustup.sh` (downloaded via the script).

cederom, 2019-04-29:
Thank you @flit will try and report back :-)

cederom, 2019-05-07:
I am finishing the hardware design.. will move into firmware development in upcoming days.. then I will get into that issue deeper.. maybe we can find a generic solution for build from sources automation for `pip` as described in https://github.com/ARMmbed/cmsis-pack-manager/issues/91 :-)

cederom, 2019-05-07:
Okay, @theotherjimmy @flit, I have quick looked at the `build-wheels.sh` and it looks very temporary code and kind of Linux'ish:
* `/bin/bash` hardcoded which is invalid because only Linux treats `bash` as `sh` while they are two different interpreters.
* Rust thing works that way too (i.e. `sh.rustup.rs > /tmp/rustup.sh` and uses `/bin/bash`).
* Hardcoded Python interpreter (i.e. `PYBIN=/opt/python/cp27-cp27m/bin`).
* Direct Python / PIP 2 invocations.
* Root directory read/write references (i.e. `/io/`, `/o`, etc).

I don't know Rust/Wheels at all, sorry. Here are some questions:
* Is this Rust some sort of bleeding edge project or has some maturity already?
* Can we use any stable release of Rust?
* Does `cargo` and `rust` take a lot disk space to build (kinda Node.JS petabytes)?
* Are we restricted to Python 2 only? Can we work with Python 3?

Any hints appreciated :-)

cederom, 2019-05-08:
Okay, I have found a system package of `rust-1.33.0` that made `cargo` work and started the build process with `python setup.py bdist_wheel` after `pip install -U -r *requirements.txt`. As simple as that, no dirty dirty `build-wheels.sh` used :-)

Duh, got `error[E0554]: #![feature] may not be used on the stable release channel` at `synstructure v0.10.1` :-(

Trying with `rust-nightly-1.35.0.20190402_1` system package.. and got `error[E0599]: no method named "resume" found for type "T" in the current scope` at `tokio v0.1.19`.

Here are questions:
* Why we don't like the stable release channel???
* Looks like this Rust is very Linux'ish / bleeding edge / self-incompatible.
* Looks like CPM was written using some particular version of Rust.
* Can we make CPM work with any stable Rust release?
* Can we stick to a known stable Rust release for CPM? That seems mandatory for platform independent automated build.
* Is there any way to install Rust as part of VirtualEnv using `pip` to get rid of unverified external dependencies, scripts and system packages?

cederom, 2019-05-08:
Thanks for feedback @theotherjimmy :-)

Looks like this `async/await` API is not yet stable (build with `rust-nightly-1.35.0.20190402_1` failed).

Do we know the schedule of `async/await` stable release?

Can we simply replace these pieces of `async/await` code with something that works with Rust Stable without breaking anything?

Would performance decrease dramatically without `async/await`?

Would replacing it now, to build on various platforms, then re-apply with stable `async/await`, cause lots of problems / work / burden? Maybe some "ifdef macro" replacing the experimental function with backward compatible code would do the trick? :-)

Do you know any way to install/run Rust from Pip/VirtualEnv? :-)

theotherjimmy, 2019-05-08:
Hei Github, I'm not a time traveler... Oh boy.

theotherjimmy, 2019-05-08:
> Do we know the schedule of async/await stable release?

I think it's in 12 weeks.

>Can we simply replace these pieces of async/await code with something that works with Rust Stable without breaking anything?

Yes.

>Would performance decrease dramatically without async/await?

No.

The limitation is that it would take time.

cederom, 2019-05-08:
> Hei Github, I'm not a time traveler... Oh boy.

I AM =)

cederom, 2019-05-08:
Can you please prepare a sample replacement code? I will do the „search-and-replace” and build with stable rust to see how that works. Huh? :-)

theotherjimmy, 2019-05-09:
@cederom It's not so simple as search and replace. The current code uses the unstable Generators feature to create an anonymous Struct that implements the Generator Trait. This transformation of "synchronous-style" program control flow into "async-style" would have to be done manually. Previously, I mentioned the files that will have to have their future implementations swapped out.

cederom, 2019-05-09:
Allright then, looks like we need to wait until the `async` stable API is out and part of Rust Release, only then we can move with works. Thank you for your time @theotherjimmy @flit you are the best ! :-)

Can I build and provide a binary for FreeBSD in the meantime using that nightly build? Would that mean providing binary for all Python versions?

theotherjimmy, 2019-05-10:
@cederom Yes, using nightly-2018-11-07 should make it possible to provide a binary wheel for python >= 2.7. cmsis-pack-manager is setup to use `cffi` instead of binding to the Python API directly, making the shared library unaware of anything python-specific.

theotherjimmy, 2019-05-11:
https://github.com/taiki-e/replace-await This may be helpful.

cederom, 2019-05-12:
Good finding! :-) Up to you @theotherjimmy if we want to convert.. don't really want to mess up your code and design :-) If you prefer to wait for stable API I understand that :-) If you want me to play with that tool out of curiosity I can try.. but I have no experience in Rust.. so I can apply and try to build.. only :-) I am on a deadline, will take a look in a few days and report back :-)

theotherjimmy, 2019-05-13:
@cederom The worry with that tool is that it kicks the ball down the road, but not all the way to stable. Since we already have a pinned version of nightly, it seems a little premature to me. I'm using this issue to keep track of many of the build-related things that affect how easy it is to port CPM to new platforms.

If you're on a deadline, then I recommend using the version of rust nightly we already pin. What's this deadline you're up against?

cederom, 2019-05-13:
Hey @theotherjimmy I have switched to macOS already so no rush, thank you for asking and concern :-) I use FreeBSD on most of my machines and it would be nice to have full MBED working over there too.. CPM is not critical as I can build the firmware which seems enough at this moment :-) We can wait for the stable rust+await as decided. In the meantime I will try to build the binary for FreeBSD based on the nightly build. Please don't put your time over here :-)  You helped enough already and the issue is brilliantly clear now, THANK YOU! :-) :-)

theotherjimmy, 2019-05-20:
@cederom I think I was able to remove part of the need for nigthly in #107. I'll be testing stable versions of Rust soon, and I'll keep this issue up to date on that progress.

theotherjimmy, 2019-05-23:
@cederom #109 makes use of stable rust, as #107 removed the actual dependency on it. Luckily, we can upgrade to any rustc >= 1.32.0 (which is a few months of releases, as we're currently at 1.34.2). That should make the rustc 1.33.0 available through the package manager of FreeBSD viable.

cederom, 2019-05-23:
awsome! thank you!! will chceck in a free moment and report back =)

cederom, 2019-05-25:
Hmm, looks like the build have some problems at `cmsis v0.1.0` test / linking with `libcmsis_cffi.so` with missing `cmsis.h` ? But the build looks different / smoother now :-)

I would also put all build dependencies in single `requrements.txt` file (including `setuptools`, `wheel`, `cargo`, etc) :-) 

```
(venv36bsd) 0xCFC2% python setup.py bdist_wheel
fatal: no tag exactly matches '7da6bc93b6140d73b489ed2af8bd86512154bd2c'
running bdist_wheel
running build
running build_py
   Compiling proc-macro2 v0.4.29
   Compiling serde v1.0.91
   Compiling parking_lot v0.7.1
   Compiling tokio-fs v0.1.6
   Compiling quick-xml v0.7.3
   Compiling tokio-reactor v0.1.9
   Compiling minidom v0.5.0
   Compiling tokio-tcp v0.1.3
   Compiling tokio-uds v0.2.5
   Compiling tokio-udp v0.1.3
   Compiling quote v0.6.12
   Compiling utils v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust/utils)
   Compiling tokio v0.1.19
   Compiling toml v0.4.10
   Compiling serde_json v1.0.39
   Compiling syn v0.15.34
   Compiling tokio-core v0.1.17
   Compiling tokio-proto v0.1.1
   Compiling tokio-rustls v0.5.0
   Compiling hyper v0.11.27
   Compiling synstructure v0.10.1
   Compiling cbindgen v0.6.8
   Compiling hyper-rustls v0.12.0
   Compiling failure_derive v0.1.5
   Compiling cmsis-cffi v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust/cmsis-cffi)
   Compiling failure v0.1.5
   Compiling pack-index v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust/pack-index)
   Compiling pdsc v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust/pdsc)
   Compiling cmsis-update v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust/cmsis-update)
   Compiling cmsis v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust)
    Finished release [optimized + debuginfo] target(s) in 7m 57s
copying dylib libcmsis_cffi.so
Traceback (most recent call last):
  File "setup.py", line 80, in <module>
    test_suite="tests"
  File "/ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages/setuptools/__init__.py", line 143, in setup
    return distutils.core.setup(**attrs)
  File "/usr/local/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/local/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/local/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages/wheel/bdist_wheel.py", line 192, in run
    self.run_command('build')
  File "/usr/local/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/local/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/usr/local/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/local/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages/milksnake/setuptools_ext.py", line 88, in run
    func(base_path=base_path, inplace=False)
  File "/ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages/milksnake/setuptools_ext.py", line 276, in build_cffi
    ffi = make_ffi()
  File "/ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages/milksnake/setuptools_ext.py", line 264, in make_ffi
    self.get_header_source(),
  File "/ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages/milksnake/setuptools_ext.py", line 231, in get_header_source
    fn = self.header_filename()
  File "setup.py", line 33, in <lambda>
    header_filename=lambda: build.find_header('cmsis.h', in_path='target')
  File "/ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages/milksnake/setuptools_ext.py", line 170, in find_header
    raise LookupError('header %r not found' % name)
LookupError: header 'cmsis.h' not found
```

cederom, 2019-05-25:
Btw. @theotherjimmy on a fresh FreeBSD system where I had to install everything from scratch and use latest pckages I have noticed that CPM became a dependency in pyOCD `0.20.0` and so pyOCD also fails to build (see https://github.com/mbedmicro/pyOCD/issues/674). This local CPM build looks more and more important and its almost here :-) :-)

cederom, 2019-05-28:
Hey @theotherjimmy is there anything I could help anyhow to fix the missing `cmsis.h`? Is it something related with librarny names as @flit noticed and reported in https://github.com/ARMmbed/cmsis-pack-manager/issues/110?

I have some issues with pyOCD on FreeBSD.. the last one that work without CPM is 0.19.0.. while there were already 3 new releases since then.. but they depend on CMP.. so I cannot really verify if problem was fixed in the new release..

Does provided code compile on your machine? What OS do you work on?

cederom, 2019-05-31:
Some updates:
* Looks like this missing `cmsis.h` is the only problem to build.
* When not building the `cmsis-cffi` build finishes with success.
* I had to update version number from `0.1.0` to `0.2.8` in order to meet `pyOCD` dependency requirements.
* Install was fine, I could then update `pyOCD` to `0.21.0` :-)

```
(venv36bsd) 0xCFC2% python setup.py install
fatal: no tag exactly matches '7da6bc93b6140d73b489ed2af8bd86512154bd2c'
running install
running bdist_egg
running egg_info
writing cmsis_pack_manager.egg-info/PKG-INFO
writing dependency_links to cmsis_pack_manager.egg-info/dependency_links.txt
writing entry points to cmsis_pack_manager.egg-info/entry_points.txt
writing requirements to cmsis_pack_manager.egg-info/requires.txt
writing top-level names to cmsis_pack_manager.egg-info/top_level.txt
reading manifest file 'cmsis_pack_manager.egg-info/SOURCES.txt'
writing manifest file 'cmsis_pack_manager.egg-info/SOURCES.txt'
installing library code to build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg
running install_lib
running build_py
   Compiling cmsis-cffi v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust/cmsis-cffi)
   Compiling cmsis v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust)
    Finished release [optimized + debuginfo] target(s) in 35.37s
creating build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg
creating build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/cmsis_pack_manager
copying build/lib/cmsis_pack_manager/_native__lib.so -> build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/cmsis_pack_manager
copying build/lib/cmsis_pack_manager/__init__.py -> build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/cmsis_pack_manager
copying build/lib/cmsis_pack_manager/pack_manager.py -> build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/cmsis_pack_manager
byte-compiling build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/cmsis_pack_manager/__init__.py to __init__.cpython-36.pyc
byte-compiling build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/cmsis_pack_manager/pack_manager.py to pack_manager.cpython-36.pyc
creating build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/EGG-INFO
copying cmsis_pack_manager.egg-info/PKG-INFO -> build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/EGG-INFO
copying cmsis_pack_manager.egg-info/SOURCES.txt -> build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/EGG-INFO
copying cmsis_pack_manager.egg-info/dependency_links.txt -> build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/EGG-INFO
copying cmsis_pack_manager.egg-info/entry_points.txt -> build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/EGG-INFO
copying cmsis_pack_manager.egg-info/not-zip-safe -> build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/EGG-INFO
copying cmsis_pack_manager.egg-info/requires.txt -> build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/EGG-INFO
copying cmsis_pack_manager.egg-info/top_level.txt -> build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/EGG-INFO
writing build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/EGG-INFO/native_libs.txt
creating 'dist/cmsis_pack_manager-0.2.8-py3.6.egg' and adding 'build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg' to it
removing 'build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg' (and everything under it)
Processing cmsis_pack_manager-0.2.8-py3.6.egg
creating /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages/cmsis_pack_manager-0.2.8-py3.6.egg
Extracting cmsis_pack_manager-0.2.8-py3.6.egg to /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages
Removing cmsis-pack-manager 0.1.1 from easy-install.pth file
Adding cmsis-pack-manager 0.2.8 to easy-install.pth file
Installing pack-manager script to /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/bin

Installed /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages/cmsis_pack_manager-0.2.8-py3.6.egg
Processing dependencies for cmsis-pack-manager==0.2.8
Searching for PyYAML==5.1
Best match: PyYAML 5.1
Adding PyYAML 5.1 to easy-install.pth file

Using /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages
Searching for milksnake==0.1.5
Best match: milksnake 0.1.5
Adding milksnake 0.1.5 to easy-install.pth file

Using /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages
Searching for appdirs==1.4.3
Best match: appdirs 1.4.3
Adding appdirs 1.4.3 to easy-install.pth file

Using /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages
Searching for cffi==1.12.3
Best match: cffi 1.12.3
Adding cffi 1.12.3 to easy-install.pth file
[... 24 lines cut ...]
Collecting prettytable (from pyocd)
  Using cached https://files.pythonhosted.org/packages/ef/30/4b0746848746ed5941f052479e7c23d2b56d174b82f4fd34a25e389831f5/prettytable-0.7.2.tar.bz2
Requirement already satisfied, skipping upgrade: pyusb<2.0,>=1.0.0b2 in /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages (from pyocd) (1.0.2)
Requirement already satisfied, skipping upgrade: intelhex<3.0,>=2.0 in /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages (from pyocd) (2.2.1)
Requirement already satisfied, skipping upgrade: pyyaml<6.0,>=5.1 in /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages (from pyocd) (5.1)
Requirement already satisfied, skipping upgrade: cmsis-pack-manager>=0.2.7 in /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages/cmsis_pack_manager-0.2.8-py3.6.egg (from pyocd) (0.2.8)
[... cut ...]
Requirement already satisfied, skipping upgrade: pycparser in /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages (from cffi>=1.6.0->milksnake>=0.1.2->cmsis-pack-manager>=0.2.7->pyocd) (2.19)
Building wheels for collected packages: prettytable
  Building wheel for prettytable (setup.py) ... done
  Stored in directory: /home/cd/.cache/pip/wheels/80/34/1c/3967380d9676d162cb59513bd9dc862d0584e045a162095606
Successfully built prettytable
Installing collected packages: prettytable, pyocd
  Found existing installation: pyocd 0.19.0
    Uninstalling pyocd-0.19.0:
      Successfully uninstalled pyocd-0.19.0
Successfully installed prettytable-0.7.2 pyocd-0.21.0
```

cederom, 2019-05-31:
Hey @theotherjimmy I have noticed recent a450f11308a14ea2fda4ca7f82df53961c83a20b tupdate to your branch @theotherjimmy, that seems to fix the `cmsis.h` issue! THANK YOU THANK YOU THANK YOU! :-)

```
(venv36bsd) 0xCFC2% git fetch jimmy
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 8 (delta 1), reused 8 (delta 1), pack-reused 0
Unpacking objects: 100% (8/8), done.
From https://github.com/theotherjimmy/cmsis-pack-manager
   7da6bc9..a450f11  use-stable -> jimmy/use-stable
(venv36bsd) 0xCFC2% git checkout jimmy/use-stable
Previous HEAD position was 7da6bc9 Use stable rust
HEAD is now at a450f11 Don't automate generation of cmsis.h
(venv36bsd) 0xCFC2% ls
LICENSE                         appveyor.yml                    cmsis_pack_manager.egg-info     rust                            test_requirements.txt
README.md                       build                           dist                            setup.cfg                       tests
apache-2.0.txt                  build-wheels.sh                 io                              setup.py
appveyor.bat                    cmsis_pack_manager              requirements.txt                setup_requirements.txt
(venv36bsd) 0xCFC2% python setup.py build
fatal: no tag exactly matches 'a450f11308a14ea2fda4ca7f82df53961c83a20b'
running build
running build_py
   Compiling syn v0.15.34
   Compiling serde v1.0.91
   Compiling serde_json v1.0.39
   Compiling synstructure v0.10.1
   Compiling failure_derive v0.1.5
   Compiling failure v0.1.5
   Compiling pack-index v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust/pack-index)
   Compiling pdsc v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust/pdsc)
   Compiling cmsis-update v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust/cmsis-update)
   Compiling cmsis-cffi v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust/cmsis-cffi)
   Compiling cmsis v0.1.0 (/ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/rust)
    Finished release [optimized + debuginfo] target(s) in 1m 39s
copying dylib libcmsis_cffi.so
generating cffi module for 'cmsis_pack_manager._native'
generating wrapper for 'cmsis_pack_manager._native'
running build_ext
building 'cmsis_pack_manager._native__lib' extension
cc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -O2 -pipe -fstack-protector -fno-strict-aliasing -fPIC -I/ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/include -I/usr/local/include/python3.6m -c build/temp.freebsd-12.0-RELEASE-p4-amd64-3.6/empty.c -o build/temp.freebsd-12.0-RELEASE-p4-amd64-3.6/build/temp.freebsd-12.0-RELEASE-p4-amd64-3.6/empty.o
cc -shared -lpthread -L/usr/local/lib -fstack-protector build/temp.freebsd-12.0-RELEASE-p4-amd64-3.6/build/temp.freebsd-12.0-RELEASE-p4-amd64-3.6/empty.o -L/usr/local/lib -lpython3.6m -o /ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/cmsis_pack_manager/_native__lib.so

(venv36bsd) 0xCFC2% python setup.py install
fatal: no tag exactly matches 'a450f11308a14ea2fda4ca7f82df53961c83a20b'
running install
running bdist_egg
running egg_info
writing cmsis_pack_manager.egg-info/PKG-INFO
writing dependency_links to cmsis_pack_manager.egg-info/dependency_links.txt
writing entry points to cmsis_pack_manager.egg-info/entry_points.txt
writing requirements to cmsis_pack_manager.egg-info/requires.txt
writing top-level names to cmsis_pack_manager.egg-info/top_level.txt
reading manifest file 'cmsis_pack_manager.egg-info/SOURCES.txt'
writing manifest file 'cmsis_pack_manager.egg-info/SOURCES.txt'
installing library code to build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg
running install_lib
running build_py
    Finished release [optimized + debuginfo] target(s) in 0.22s
copying dylib libcmsis_cffi.so
generating cffi module for 'cmsis_pack_manager._native'
already up-to-date
generating wrapper for 'cmsis_pack_manager._native'
running build_ext
building 'cmsis_pack_manager._native__lib' extension
cc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -O2 -pipe -fstack-protector -fno-strict-aliasing -fPIC -I/ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/include -I/usr/local/include/python3.6m -c build/temp.freebsd-12.0-RELEASE-p4-amd64-3.6/empty.c -o build/temp.freebsd-12.0-RELEASE-p4-amd64-3.6/build/temp.freebsd-12.0-RELEASE-p4-amd64-3.6/empty.o
cc -shared -lpthread -L/usr/local/lib -fstack-protector build/temp.freebsd-12.0-RELEASE-p4-amd64-3.6/build/temp.freebsd-12.0-RELEASE-p4-amd64-3.6/empty.o -L/usr/local/lib -lpython3.6m -o /ztuff/cederom/work/CeDeROM/lab/embedded/arm/mbed/cederom-cmsis-pack-manager.git/cmsis_pack_manager/_native__lib.so
creating build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg
creating build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/cmsis_pack_manager
copying build/lib/cmsis_pack_manager/_native__lib.so -> build/bdist.freebsd-12.0-RELEASE-p4-amd64/egg/cmsis_pack_manager
[... 47 lines cut ...]
Adding cffi 1.12.3 to easy-install.pth file

Using /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages
Searching for pycparser==2.19
Best match: pycparser 2.19
Adding pycparser 2.19 to easy-install.pth file

Using /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages
Finished processing dependencies for cmsis-pack-manager==0.2.8

(venv36bsd) 0xCFC2% pip show cmsis-pack-manager
Name: cmsis-pack-manager
Version: 0.2.8
Summary: UNKNOWN
Home-page: UNKNOWN
Author: UNKNOWN
Author-email: UNKNOWN
License: UNKNOWN
Location: /ztuff/cederom/work/CeDeROM/lab/python/venv36bsd/lib/python3.6/site-packages/cmsis_pack_manager-0.2.8-py3.6-freebsd-12.0-RELEASE-p4-amd64.egg
Requires: appdirs, milksnake, pyyaml
Required-by: pyocd

```

When the release is done and the tag is set I guess the version would also adjust itself :-)

theotherjimmy, 2019-05-31:
I'm glad it's working for you. I'll tag a new release ASAP.

theotherjimmy, 2019-05-31:
Released the fix. Closing.

closed 2019-05-31 by theotherjimmy

cederom, 2019-05-31:
TANK U SIR!!! =)