pyocd/cmsis-pack-manager #101: Add proxy support

flit, 2019-05-08:
This is a request to add support for the `http_proxy` environment variable so that the index and packs can be downloaded through a proxy server.

jeromecoutant, 2019-05-09:
@rgf97

theotherjimmy, 2019-05-09:
Good debugging @jeromecoutant Thanks for all the detail. I actually use a `GET` from my API, I just need to figure out how to let the OS do the DNS (like everyone else...)

rmu75, 2019-06-03:
The issue is not the DNS, the issue is not using the proxy. In my environment, DNS resolution of www.keil.com works just fine, but http requests need to go through the proxy. pyocd pack --install fails with "Connection refused (os error 111)".

flit, 2019-06-03:
There are probably multiple proxy-related issues here, depending on the network setup. The error reported by @jeromecoutant was "No such host is known. (os error 11001)", which indicates a DNS failure.

rmu75, 2019-06-03:
> There are probably multiple proxy-related issues here, depending on the network setup.

I don't think so. In the error reported by @jeromecoutant the proxy is perfectly reachable and hostname of proxy resolves, so there are no DNS issues.

I'm not really fluent in rust and don't know the library landscape, so I can't really check in the source, nevertheless, strace(1)ing pyocd suggest http_proxy / https_proxy environment variables are ignored, iow no proxy functionality is implemented.

theotherjimmy, 2019-10-07:
@jeromecoutant @rmu75 Would you be willing to run a test build from my branch https://github.com/theotherjimmy/cmsis-pack-manager/tree/use-reqwest to confirm proxy support in your setups?

rmu75, 2019-11-13:
Sorry for the late reply, very busy atm. Of course I am willing to test this. Thanks.

flit, 2020-11-24:
Closing this issue, it should be fixed. CPM now uses reqwest, and calls `.use_sys_proxy()`.

If someone encounters the problem again, please reopen.

closed 2020-11-24 by flit