pyocd/cmsis-pack-manager #106: macOS - ERROR: Could not find a version that satisfies the requirement cmsis-pack-manager>=0.2.6 (from pyocd) (from versions: 0.1.0)

jhgorse, 2019-05-18:
When installing from venv or otherwise:

```
ERROR: Could not find a version that satisfies the requirement cmsis-pack-manager>=0.2.6 (from pyocd) (from versions: 0.1.0)
ERROR: No matching distribution found for cmsis-pack-manager>=0.2.6 (from pyocd)
```

Trying to build cmsis-pack-manager results in 

```
venv] jhg@i7joe:/private/tmp/cmsis-pack-manager                                                                                                       > python setup.py build
warning: no previously-included files found matching 'setup.pyc'
warning: no previously-included files matching 'yacctab.*' found under directory 'tests'
warning: no previously-included files matching 'lextab.*' found under directory 'tests'
warning: no previously-included files matching 'yacctab.*' found under directory 'examples'
warning: no previously-included files matching 'lextab.*' found under directory 'examples'
zip_safe flag not set; analyzing archive contents...
pycparser.ply.lex: module references __file__
pycparser.ply.lex: module MAY be using inspect.getsourcefile
pycparser.ply.yacc: module references __file__
pycparser.ply.yacc: module MAY be using inspect.getsourcefile
pycparser.ply.yacc: module MAY be using inspect.stack
pycparser.ply.ygen: module references __file__

Installed /private/tmp/cmsis-pack-manager/.eggs/pycparser-2.19-py2.7.egg
running build
running build_py
error: [Errno 2] No such file or directory
Exit 1
```

Cheers,
Joe

theotherjimmy, 2019-05-20:
@jhgorse, cmsis-pack-manager is mostly written in rust (for speed; see #18 for details there). To avoid required the user to install the rust compiler, we build binary wheels (as part of the CI) and release only the binary wheels. see #91 and #97 for longer discussions on what exactly is require.

OTOH, we release for macOS (I have a mac as well), and you should be able to download a compatible binary wheel. Could you try explicitly installing the macOS version from here: https://github.com/ARMmbed/cmsis-pack-manager/releases/tag/v0.2.6 and post the error that you get?

flit, 2019-05-20:
Hi @jhgorse , what macOS version are you running?

flit, 2019-10-05:
@jhgorse Is this issue resolved? Can it be closed?

jhgorse, 2019-10-07:
@flit Latest publicly released macOS.

We can close for now as I don't have time to revisit.

Thank you for getting back to me! See you! =)

closed 2019-10-07 by jhgorse