pyocd/cmsis-pack-manager #117: cmsis-pack-manager 0.1.1

rmu75, 2019-07-30:
on a raspberry pi running ubuntu 18.04.2 "pip3 install pyocd" complains about "cmsis-pack-manager 0.1.1":

```
ERROR: pyocd 0.21.0 has requirement cmsis-pack-manager>=0.2.7, but you'll have cmsis-pack-manager 0.1.1 which is incompatible.
Installing collected packages: cmsis-pack-manager
  Found existing installation: cmsis-pack-manager 0.1.1
    Uninstalling cmsis-pack-manager-0.1.1:
      Successfully uninstalled cmsis-pack-manager-0.1.1
Successfully installed cmsis-pack-manager-0.1.1
```

upon investigation, this
https://github.com/ARMmbed/cmsis-pack-manager/blob/e26f87d797715aa2f5111b6333e0908dea8ece80/setup.py#L55

seems to be the cause.

after downloading 0.2.8 source tarball from pypi, changing "0.1.1" to "0.2.8", building cmsis-pack-manager and installation in a new virtualenv, installation of pyocd finally succeeds.

theotherjimmy, 2019-07-30:
Yeah, that makes sense. I can't upload new tarballs for 0.2.7 and 0.2.8, and this requires changes to setup.py. I don't know why I missed that.

theotherjimmy, 2019-07-30:
I'm working on #116 and a fix for that: packaging a "version.txt" that includes the version info from the release.

flit, 2019-07-30:
You could use the method we use for pyOCD where the version is automatically determined by the latest git tag, with the `setuptools_scm` package added to the `setup_requires` key for `setup()`. See [`setup.py`](https://github.com/mbedmicro/pyOCD/blob/master/setup.py#L38). 

The `setuptools_scm_git_archive` package enables this to work with git archives downloaded from GitHub (as opposed to "live" clones). For this, you also need to add the [`.git_archival.txt`](https://github.com/mbedmicro/pyOCD/blob/master/.git_archival.txt) file.

theotherjimmy, 2019-07-30:
That's a good suggestion. I'll look into that over #118

theotherjimmy, 2019-07-30:
That seems like a fantastic solution, and should let me remove many of my version hacks.

theotherjimmy, 2019-07-30:
Excellent suggestion did I do #119 correctly @flit?

flit, 2019-07-30:
@theotherjimmy Looks perfect! Sorry I forgot some of the other bits like the `.gitattributes` change and `use_scm_version` key.

theotherjimmy, 2019-07-30:
No worries. I read the documentation and looked through your setup.py

theotherjimmy, 2019-11-18:
I released 0.2.9 with the changes Chris suggested. @rmu75 does this issue remain with the newer release?

closed 2021-04-03 by flit