pyocd/cmsis-pack-manager #121: Re-direct support missing from pack downloader?

JanneKiiskila, 2019-10-03:
If you try to update the cmsis packs in Mbed OS with `python project.py -m STM32F7xx --update-packs` it will start downloading the pack files. 

For a few ST packs, you will get the following error note:

```
34:25.840 ERRO download of "https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc" failed: unexpected end of file
Oct 02 18:34:25.840 ERRO download of "https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc" failed: unexpected end of file
Oct 02 18:34:25.840 ERRO download of "https://www.keil.com/pack/Keil.STM32G0xx_DFP.pdsc" failed: unexpected end of file
```

If you do a curl on those addresses, you will find out that actually - they are not the right addresses - you will get a re-direct instead.

For example (preceded by a very long header):
```
</head><body>
<h1>Found</h1><p>The document has moved <a href="http://sadevicepacksprodus.blob.core.windows.net/pdsc/Keil.STM32F7xx_DFP.pdsc">here</a>.</p>
```

Seems one should use something for the download, which supports redirects?
I'm assuming the downloader is actually here; 

https://github.com/ARMmbed/cmsis-pack-manager/blob/d522e704ac2ca2f116c4e1fde73357dee858218e/rust/cmsis-update/src/download.rs#L144-L170

and it seems to use `futures::Stream;`.

theotherjimmy, 2019-10-03:
Nope. We support redirects. _All_ of the keil.com pdcs redirect. These are not special. And we download the others. Perhaps they do not use a 300 for the redirect.

---------------

The linked code does not implement downloading, it calls the actual download function across a stream of files. What does `futures::Stream` have to do with all of this?

JanneKiiskila, 2019-10-03:
They give a 302. 

```
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head><script type="text/javascript">window.NREUM||(NREUM={});NREUM.info = {"beacon":"bam.nr-data.net","errorBeacon":"bam.nr-data.net","licenseKey":"c4afc5d5b0","applicationID":"38150566","transactionName":"Y1FVbUUFDBAAU0xeXVobZE1WEBcQIl9cUh0HBAU=","queueTime":0,"applicationTime":1,"ttGuid":"ED76BAEBF58981FC","agent":""}</script><script type="text/javascript">window.NREUM||(NREUM={}),__nr_require=function(e,n,t){function r(t){if(!n[t]){var o=n[t]={exports:{}};e[t][0].call(o.exports,function(n){var o=e[t][1][n];return r(o||n)},o,o.exports)}return n[t].exports}if("function"==typeof __nr_require)return __nr_require;for(var o=0;o<t.length;o++)r(t[o]);return r}({1:[function(e,n,t){function r(){}function o(e,n,t){return function(){return i(e,[c.now()].concat(u(arguments)),n?null:this,t),n?void 0:this}}var i=e("handle"),a=e(3),u=e(4),f=e("ee").get("tracer"),c=e("loader"),s=NREUM;"undefined"==typeof window.newrelic&&(newrelic=s);var p=["setPageViewName","setCustomAttribute","setErrorHandler","finished","addToTrace","inlineHit","addRelease"],d="api-",l=d+"ixn-";a(p,function(e,n){s[n]=o(d+n,!0,"api")}),s.addPageAction=o(d+"addPageAction",!0),s.setCurrentRouteName=o(d+"routeName",!0),n.exports=newrelic,s.interaction=function(){return(new r).get()};var m=r.prototype={createTracer:function(e,n){var t={},r=this,o="function"==typeof n;return i(l+"tracer",[c.now(),e,t],r),function(){if(f.emit((o?"":"no-")+"fn-start",[c.now(),r,o],t),o)try{return n.apply(this,arguments)}catch(e){throw f.emit("fn-err",[arguments,this,e],t),e}finally{f.emit("fn-end",[c.now()],t)}}}};a("actionText,setName,setAttribute,save,ignore,onEnd,getContext,end,get".split(","),function(e,n){m[n]=o(l+n)}),newrelic.noticeError=function(e,n){"string"==typeof e&&(e=new Error(e)),i("err",[e,c.now(),!1,n])}},{}],2:[function(e,n,t){function r(e,n){if(!o)return!1;if(e!==o)return!1;if(!n)return!0;if(!i)return!1;for(var t=i.split("."),r=n.split("."),a=0;a<r.length;a++)if(r[a]!==t[a])return!1;return!0}var o=null,i=null,a=/Version\/(\S+)\s+Safari/;if(navigator.userAgent){var u=navigator.userAgent,f=u.match(a);f&&u.indexOf("Chrome")===-1&&u.indexOf("Chromium")===-1&&(o="Safari",i=f[1])}n.exports={agent:o,version:i,match:r}},{}],3:[function(e,n,t){function r(e,n){var t=[],r="",i=0;for(r in e)o.call(e,r)&&(t[i]=n(r,e[r]),i+=1);return t}var o=Object.prototype.hasOwnProperty;n.exports=r},{}],4:[function(e,n,t){function r(e,n,t){n||(n=0),"undefined"==typeof t&&(t=e?e.length:0);for(var r=-1,o=t-n||0,i=Array(o<0?0:o);++r<o;)i[r]=e[n+r];return i}n.exports=r},{}],5:[function(e,n,t){n.exports={exists:"undefined"!=typeof window.performance&&window.performance.timing&&"undefined"!=typeof window.performance.timing.navigationStart}},{}],ee:[function(e,n,t){function r(){}function o(e){function n(e){return e&&e instanceof r?e:e?f(e,u,i):i()}function t(t,r,o,i){if(!d.aborted||i){e&&e(t,r,o);for(var a=n(o),u=v(t),f=u.length,c=0;c<f;c++)u[c].apply(a,r);var p=s[y[t]];return p&&p.push([b,t,r,a]),a}}function l(e,n){h[e]=v(e).concat(n)}function m(e,n){var t=h[e];if(t)for(var r=0;r<t.length;r++)t[r]===n&&t.splice(r,1)}function v(e){return h[e]||[]}function g(e){return p[e]=p[e]||o(t)}function w(e,n){c(e,function(e,t){n=n||"feature",y[t]=n,n in s||(s[n]=[])})}var h={},y={},b={on:l,addEventListener:l,removeEventListener:m,emit:t,get:g,listeners:v,context:n,buffer:w,abort:a,aborted:!1};return b}function i(){return new r}function a(){(s.api||s.feature)&&(d.aborted=!0,s=d.backlog={})}var u="nr@context",f=e("gos"),c=e(3),s={},p={},d=n.exports=o();d.backlog=s},{}],gos:[function(e,n,t){function r(e,n,t){if(o.call(e,n))return e[n];var r=t();if(Object.defineProperty&&Object.keys)try{return Object.defineProperty(e,n,{value:r,writable:!0,enumerable:!1}),r}catch(i){}return e[n]=r,r}var o=Object.prototype.hasOwnProperty;n.exports=r},{}],handle:[function(e,n,t){function r(e,n,t,r){o.buffer([e],r),o.emit(e,n,t)}var o=e("ee").get("handle");n.exports=r,r.ee=o},{}],id:[function(e,n,t){function r(e){var n=typeof e;return!e||"object"!==n&&"function"!==n?-1:e===window?0:a(e,i,function(){return o++})}var o=1,i="nr@id",a=e("gos");n.exports=r},{}],loader:[function(e,n,t){function r(){if(!E++){var e=x.info=NREUM.info,n=l.getElementsByTagName("script")[0];if(setTimeout(s.abort,3e4),!(e&&e.licenseKey&&e.applicationID&&n))return s.abort();c(y,function(n,t){e[n]||(e[n]=t)}),f("mark",["onload",a()+x.offset],null,"api");var t=l.createElement("script");t.src="https://"+e.agent,n.parentNode.insertBefore(t,n)}}function o(){"complete"===l.readyState&&i()}function i(){f("mark",["domContent",a()+x.offset],null,"api")}function a(){return O.exists&&performance.now?Math.round(performance.now()):(u=Math.max((new Date).getTime(),u))-x.offset}var u=(new Date).getTime(),f=e("handle"),c=e(3),s=e("ee"),p=e(2),d=window,l=d.document,m="addEventListener",v="attachEvent",g=d.XMLHttpRequest,w=g&&g.prototype;NREUM.o={ST:setTimeout,SI:d.setImmediate,CT:clearTimeout,XHR:g,REQ:d.Request,EV:d.Event,PR:d.Promise,MO:d.MutationObserver};var h=""+location,y={beacon:"bam.nr-data.net",errorBeacon:"bam.nr-data.net",agent:"js-agent.newrelic.com/nr-1130.min.js"},b=g&&w&&w[m]&&!/CriOS/.test(navigator.userAgent),x=n.exports={offset:u,now:a,origin:h,features:{},xhrWrappable:b,userAgent:p};e(1),l[m]?(l[m]("DOMContentLoaded",i,!1),d[m]("load",r,!1)):(l[v]("onreadystatechange",o),d[v]("onload",r)),f("mark",["firstbyte",u],null,"api");var E=0,O=e(5)},{}]},{},["loader"]);</script>
<title>302 Found</title>
</head><body>
<h1>Found</h1><p>The document has moved <a href="http://sadevicepacksprodus.blob.core.windows.net/pdsc/Keil.STM32F7xx_DFP.pdsc">here</a>.</p>
```

theotherjimmy, 2019-10-03:
> Well, actually they give likely just 200 OK as the error code and then a page with .

That sounds like a 200.

theotherjimmy, 2019-10-03:
Yeah, it's actually really hard to write a command line tool that takes a 200 OK and realizes that it's actually a redirect. Also note that the error might have nothing to do with this, as it's about the file ending too soon.

theotherjimmy, 2019-10-03:
```
❯❯ curl https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc -D header
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head><script type="text/javascript">window.NREUM||(NREUM={});NREUM.info = {"beacon":"bam.nr-data.net","errorBeacon":"bam.nr-data.net","licenseKey":"c4afc5d5b0","applicationID":"38150566","transactionName":"Y1FVbUUFDBAAU0xeXVobZE1WEBcQIl9cUh0HBAU=","queueTime":0,"applicationTime":1,"ttGuid":"93D726A322488A18","agent":""}</script><script type="text/javascript">window.NREUM||(NREUM={}),__nr_require=function(e,n,t){function r(t){if(!n[t]){var o=n[t]={exports:{}};e[t][0].call(o.exports,function(n){var o=e[t][1][n];return r(o||n)},o,o.exports)}return n[t].exports}if("function"==typeof __nr_require)return __nr_require;for(var o=0;o<t.length;o++)r(t[o]);return r}({1:[function(e,n,t){function r(){}function o(e,n,t){return function(){return i(e,[c.now()].concat(u(arguments)),n?null:this,t),n?void 0:this}}var i=e("handle"),a=e(3),u=e(4),f=e("ee").get("tracer"),c=e("loader"),s=NREUM;"undefined"==typeof window.newrelic&&(newrelic=s);var p=["setPageViewName","setCustomAttribute","setErrorHandler","finished","addToTrace","inlineHit","addRelease"],d="api-",l=d+"ixn-";a(p,function(e,n){s[n]=o(d+n,!0,"api")}),s.addPageAction=o(d+"addPageAction",!0),s.setCurrentRouteName=o(d+"routeName",!0),n.exports=newrelic,s.interaction=function(){return(new r).get()};var m=r.prototype={createTracer:function(e,n){var t={},r=this,o="function"==typeof n;return i(l+"tracer",[c.now(),e,t],r),function(){if(f.emit((o?"":"no-")+"fn-start",[c.now(),r,o],t),o)try{return n.apply(this,arguments)}catch(e){throw f.emit("fn-err",[arguments,this,e],t),e}finally{f.emit("fn-end",[c.now()],t)}}}};a("actionText,setName,setAttribute,save,ignore,onEnd,getContext,end,get".split(","),function(e,n){m[n]=o(l+n)}),newrelic.noticeError=function(e,n){"string"==typeof e&&(e=new Error(e)),i("err",[e,c.now(),!1,n])}},{}],2:[function(e,n,t){function r(e,n){if(!o)return!1;if(e!==o)return!1;if(!n)return!0;if(!i)return!1;for(var t=i.split("."),r=n.split("."),a=0;a<r.length;a++)if(r[a]!==t[a])return!1;return!0}var o=null,i=null,a=/Version\/(\S+)\s+Safari/;if(navigator.userAgent){var u=navigator.userAgent,f=u.match(a);f&&u.indexOf("Chrome")===-1&&u.indexOf("Chromium")===-1&&(o="Safari",i=f[1])}n.exports={agent:o,version:i,match:r}},{}],3:[function(e,n,t){function r(e,n){var t=[],r="",i=0;for(r in e)o.call(e,r)&&(t[i]=n(r,e[r]),i+=1);return t}var o=Object.prototype.hasOwnProperty;n.exports=r},{}],4:[function(e,n,t){function r(e,n,t){n||(n=0),"undefined"==typeof t&&(t=e?e.length:0);for(var r=-1,o=t-n||0,i=Array(o<0?0:o);++r<o;)i[r]=e[n+r];return i}n.exports=r},{}],5:[function(e,n,t){n.exports={exists:"undefined"!=typeof window.performance&&window.performance.timing&&"undefined"!=typeof window.performance.timing.navigationStart}},{}],ee:[function(e,n,t){function r(){}function o(e){function n(e){return e&&e instanceof r?e:e?f(e,u,i):i()}function t(t,r,o,i){if(!d.aborted||i){e&&e(t,r,o);for(var a=n(o),u=v(t),f=u.length,c=0;c<f;c++)u[c].apply(a,r);var p=s[y[t]];return p&&p.push([b,t,r,a]),a}}function l(e,n){h[e]=v(e).concat(n)}function m(e,n){var t=h[e];if(t)for(var r=0;r<t.length;r++)t[r]===n&&t.splice(r,1)}function v(e){return h[e]||[]}function g(e){return p[e]=p[e]||o(t)}function w(e,n){c(e,function(e,t){n=n||"feature",y[t]=n,n in s||(s[n]=[])})}var h={},y={},b={on:l,addEventListener:l,removeEventListener:m,emit:t,get:g,listeners:v,context:n,buffer:w,abort:a,aborted:!1};return b}function i(){return new r}function a(){(s.api||s.feature)&&(d.aborted=!0,s=d.backlog={})}var u="nr@context",f=e("gos"),c=e(3),s={},p={},d=n.exports=o();d.backlog=s},{}],gos:[function(e,n,t){function r(e,n,t){if(o.call(e,n))return e[n];var r=t();if(Object.defineProperty&&Object.keys)try{return Object.defineProperty(e,n,{value:r,writable:!0,enumerable:!1}),r}catch(i){}return e[n]=r,r}var o=Object.prototype.hasOwnProperty;n.exports=r},{}],handle:[function(e,n,t){function r(e,n,t,r){o.buffer([e],r),o.emit(e,n,t)}var o=e("ee").get("handle");n.exports=r,r.ee=o},{}],id:[function(e,n,t){function r(e){var n=typeof e;return!e||"object"!==n&&"function"!==n?-1:e===window?0:a(e,i,function(){return o++})}var o=1,i="nr@id",a=e("gos");n.exports=r},{}],loader:[function(e,n,t){function r(){if(!E++){var e=x.info=NREUM.info,n=l.getElementsByTagName("script")[0];if(setTimeout(s.abort,3e4),!(e&&e.licenseKey&&e.applicationID&&n))return s.abort();c(y,function(n,t){e[n]||(e[n]=t)}),f("mark",["onload",a()+x.offset],null,"api");var t=l.createElement("script");t.src="https://"+e.agent,n.parentNode.insertBefore(t,n)}}function o(){"complete"===l.readyState&&i()}function i(){f("mark",["domContent",a()+x.offset],null,"api")}function a(){return O.exists&&performance.now?Math.round(performance.now()):(u=Math.max((new Date).getTime(),u))-x.offset}var u=(new Date).getTime(),f=e("handle"),c=e(3),s=e("ee"),p=e(2),d=window,l=d.document,m="addEventListener",v="attachEvent",g=d.XMLHttpRequest,w=g&&g.prototype;NREUM.o={ST:setTimeout,SI:d.setImmediate,CT:clearTimeout,XHR:g,REQ:d.Request,EV:d.Event,PR:d.Promise,MO:d.MutationObserver};var h=""+location,y={beacon:"bam.nr-data.net",errorBeacon:"bam.nr-data.net",agent:"js-agent.newrelic.com/nr-1130.min.js"},b=g&&w&&w[m]&&!/CriOS/.test(navigator.userAgent),x=n.exports={offset:u,now:a,origin:h,features:{},xhrWrappable:b,userAgent:p};e(1),l[m]?(l[m]("DOMContentLoaded",i,!1),d[m]("load",r,!1)):(l[v]("onreadystatechange",o),d[v]("onload",r)),f("mark",["firstbyte",u],nu [... cut ...]
</body></html>                                                                                                                                                 [ 0s512 | Oct 03 08:50AM ]
[I] ~/s/a/b/p/meta-pelion-edge❯❯ cat header
HTTP/1.1 302 Found
Content-Type: text/html
Location: http://sadevicepacksprodus.blob.core.windows.net/pdsc/Keil.STM32F7xx_DFP.pdsc
Server: Microsoft-IIS/8.5
X-Powered-By: ASP.NET
X-UA-Compatible: IE=EDGE
Date: Thu, 03 Oct 2019 13:50:18 GMT
Connection: close
Content-Length: 5683
```
So it's probably not the redirect code.

theotherjimmy, 2019-10-03:
Seems like this is a bug:
```
curl http://sadevicepacksprodus.blob.core.windows.net/pdsc/Keil.STM32F7xx_DFP.pdsc > foo.txt
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  335k  100  335k    0     0   786k      0 --:--:-- --:--:-- --:--:--  786k
```

theotherjimmy, 2019-10-03:
In a debug build, it looks like it does not receive the redirect:
```
Oct 03 09:21:25.244 DEBG Starting GET of https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc
Oct 03 09:21:25.246 ERRO download of "https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc" failed: unexpected end of file
```

If it worked it would look like:
```
Oct 03 09:21:25.227 DEBG Starting GET of http://mcuxpresso.nxp.com/cmsis_pack/repo/NXP.LPCXpresso54102_BSP.pdsc
Oct 03 09:21:25.227 DEBG Redirecting from http://mcuxpresso.nxp.com/cmsis_pack/repo/NXP.LPCXpresso54102_BSP.pdsc to https://mcuxpresso.nxp.com/cmsis_pack/repo/NXP.LPCXpresso54102_BSP.pdsc
Oct 03 09:21:25.227 DEBG Starting GET of https://mcuxpresso.nxp.com/cmsis_pack/repo/NXP.LPCXpresso54102_BSP.pdsc
```

jkrech, 2019-10-04:
I am unable to reproduce the problem:

```
jkrech@E123565 MINGW64 /c/tmp/test
$ curl http://sadevicepacksprodus.blob.core.windows.net/pdsc/Keil.STM32F7xx_DFP.pdsc -D header
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0<?xml version="1.0" encoding="UTF-8"?>

<package schemaVersion="1.6.0" xmlns:xs="http://www.w3.org/2001/XMLSchema-instance" xs:noNamespaceSchemaLocation="PACK.xsd">
  <vendor>Keil</vendor>
  <name>STM32F7xx_DFP</name>
  <description>STMicroelectronics STM32F7 Series Device Support, Drivers and Examples</description>
  <url>https://www.keil.com/pack/</url>

  <releases>
    <release version="2.12.0" date="2019-07-17">

```

From my perspective (I am just uploading a pack through a Web Interface) there is nothing special about the Keil.STM32MH7xx_DFP.pdsc. 
For better download performance around the world we  host the files on Azure. I do expect all pdsc files at www.keil.com/pack to redirect.

Once you instruct curl to follow the redirect I do see the pdsc file being properly downloaded on my machine (</package>) .

Sorry if I cannot provide any more insights. Please note that I also validated that the MDK Pack Installer would be able to download successfully.

theotherjimmy, 2019-10-04:
I think you're right; I think it's something going wrong in cmsis-pack-manager. It's odd that it only affects these 3 packs, and that it's consistent.

JanneKiiskila, 2019-10-04:
Perhaps a strange idea, but what if we replaced that download code by a call to "curl" itself, i.e. `curl <Uri> -L ><PathBuf>`, if you see what I mean. Seems `curl` can handle the redirects, then why not use it?

flit, 2019-10-04:
Because that would be a nasty dependency to manage.

theotherjimmy, 2019-10-04:
@JanneKiiskila all this talk about handling redirects is a red herring, as this fails before invoking any redirection code.

JanneKiiskila, 2019-10-04:
Yep, that's what you get when a noob starts looking at the rust code we have in this pack without any real understanding. Title and description modified.

JanneKiiskila, 2019-10-04:
The older I get, the more I know what I don't know.

theotherjimmy, 2019-10-04:
Thanks for changing the title. I'm trying to root cause the issue here. This is weird that curl does not have the same issue.

theotherjimmy, 2019-10-04:
I have ruled out concurrency, as I removed concurrent downloads in a test build without affecting the problem behavior.

JanneKiiskila, 2019-10-04:
Thanks for looking into this!

theotherjimmy, 2019-10-04:
Snippet from strace:
```
Oct 04 14:22:27.717 DEBG Starting GET of https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc
recvfrom(32, "", 18437, 0, NULL, NULL)  = 0
epoll_ctl(27, EPOLL_CTL_DEL, 32, 0x7ffd8dc89a70) = 0
close(32)  
```
A `0` from recvfrom indicates that "If no messages are available to be received and the peer has performed an orderly shutdown, recvfrom() shall return 0". 

I'm thinking that CPM might be sending an invalid GET to the server and it's responds by shutting down the connection.

theotherjimmy, 2019-10-04:
I just traced a clean download of everything and it looks like all of these are https keil.com packs being, and these failing to download packs are the only packs that use https from keil.com.

JanneKiiskila, 2019-10-04:
Why are there so many Starting GET of for the same file? Should not one be enough, or is the debug print just slightly misleading?

theotherjimmy, 2019-10-04:
It looks like this might be a TLS cyphersuite miss-match. CMP uses Rustls for TLS, and as such does not support TLS < 1.2. Using Openssl, I was able to figure out that www.keil.com supports the following cyphersuites:
```
tls1_2:	ECDHE-RSA-AES256-SHA384
tls1_2:	ECDHE-RSA-AES128-SHA256
tls1_2:	ECDHE-RSA-AES256-SHA
tls1_2:	ECDHE-RSA-AES128-SHA
tls1_2:	AES256-SHA256
tls1_2:	AES128-SHA256
tls1_2:	AES256-SHA
tls1_2:	AES128-SHA
```

This might be a TLS cyphersuite mismatch.

JanneKiiskila, 2019-10-04:
If `curl` is a bad idea, could we use a Rust -based solution? Seems Rust has a HTTP download crate called [`reqwest`](https://rust-lang-nursery.github.io/rust-cookbook/web/clients/download.html).

theotherjimmy, 2019-10-04:
@JanneKiiskila I'm not sure. I was thinking the same thing: why are we downloading the same thing 3-4 times? I'm going to put that aside for now and try to track down this TLS issue.

JanneKiiskila, 2019-10-04:
I'm just wondering could it be an DDoS protection mechanism, i.e. it blocks the connection because it thinks someone is doing too much at same time?

theotherjimmy, 2019-10-04:
It could be a DDoS protection mechanism.

I tried the Rustls example with www.keil.com:
```
❯ cargo run --example tlsclient -- --http expired.badssl.com
warning: /home/jimmy/src/ctz/rustls/rustls/Cargo.toml: file found to be present in multiple build targets: /home/jimmy/src/ctz/rustls/rustls/tests/benchmarks.rs
    Finished dev [unoptimized + debuginfo] target(s) in 0.04s
     Running `target/debug/examples/tlsclient --http expired.badssl.com`
TLS error: WebPKIError(CertExpired)
Connection closed
                                                                [ 0s196 | Oct 04 03:11PM ]
[I] ~/s/c/rustls❯❯ cargo run --example tlsclient -- --http www.keil.com
warning: /home/jimmy/src/ctz/rustls/rustls/Cargo.toml: file found to be present in multiple build targets: /home/jimmy/src/ctz/rustls/rustls/tests/benchmarks.rs
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `target/debug/examples/tlsclient --http www.keil.com`
EOF
Connection closed
                                                                [ 0s356 | Oct 04 03:11PM ]
[I] ~/s/c/rustls❯❯ cargo run --example tlsclient -- --http mozilla-modern.badssl.com
warning: /home/jimmy/src/ctz/rustls/rustls/Cargo.toml: file found to be present in multiple build targets: /home/jimmy/src/ctz/rustls/rustls/tests/benchmarks.rs
    Finished dev [unoptimized + debuginfo] target(s) in 0.11s
     Running `target/debug/examples/tlsclient --http mozilla-modern.badssl.com`
HTTP/1.1 200 OK
Server: nginx/1.10.3 (Ubuntu)
Date: Fri, 04 Oct 2019 20:16:32 GMT
Content-Type: text/html
Content-Length: 644
Last-Modified: Wed, 11 Sep 2019 15:20:49 GMT
Connection: close
ETag: "5d7910d1-284"
Strict-Transport-Security: max-age=15768000
Cache-Control: no-store
Accept-Ranges: bytes

<!DOCTYPE html>
<html>
<head>
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <link rel="shortcut icon" href="/icons/favicon-green.ico"/>
  <link rel="apple-touch-icon" href="/icons/icon-green.png"/>
  <title>mozilla-modern.badssl.com</title>
  <link rel="stylesheet" href="/style.css">
  <style>body { background: green; }</style>
</head>
<body>
<div id="content">
  <h1>
    mozilla-modern.<br>badssl.com
  </h1>
</div>

<div id="footer">
  This site uses the Mozilla &ldquo;<a href="https://wiki.mozilla.org/Security/Server_Side_TLS#Modern_compatibility">Modern</a>&rdquo; TLS configuration.
</div>

</body>
</html>
Plaintext read error: Custom { kind: ConnectionAborted, error: "CloseNotify alert received" }
Connection closed
```
It's the middle one. Other commands provided for context.

theotherjimmy, 2019-10-04:
@JanneKiiskila Good digging in libraries. I'm using `hyper` + `rustls`. `reqwest` is built on top of `hyper` and a i would be happy to use `reqwest` if it fixes the problem. Otherwise, I see no reason to switch, and I see a small downside to adding more dependencies.

JanneKiiskila, 2019-10-04:
A multi-curl should theoretically then also fail, if there was a DDoS mechanism there, but it seems to work. At least this;

```
curl https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc -L >1.txt & curl https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc -L >2.txt & curl https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc -L >3.txt & curl https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc -L >4.txt & curl https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc -L >5.txt & curl https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc -L >6.txt
```

Got all `1...6.txt` files down and they are all the same, with the proper content.

flit, 2019-10-04:
have you tried capturing the tls handshake with wireshark?

theotherjimmy, 2019-10-04:
@flit i have not tried capturing the tls handshake with wireshark yet. Good suggestion, I'll give that a shot.

flit, 2019-10-04:
i'm also trying it, but you probably have more experience 😄

flit, 2019-10-04:
Here's a screenshot of the capture of one of the requests:

[image]

Looks like the client sends the TLS hello message, then the server drops the connection without a response. Not very helpful in debugging the why…

JanneKiiskila, 2019-10-05:
I am getting errors downloading these, too:

```
Oct 05 04:01:13.321 ERRO download of "http://packs.download.atmel.com/Atmel.SAM3X_DFP.1.0.50.pack" failed: Connection reset by peer (os error 104)
Oct 05 08:19:06.241 ERRO download of "http://packs.download.atmel.com/Atmel.SAME54_DFP.1.1.134.pack" failed: Connection reset by peer (os error 104)
```

I can see from Wireshark it tries to resolve those names, redirect comes, but typically it all ends up with RST. Almost as if the client side would have sent something the servers do not like?

Downloading the same file with `curl` works again.

```
 curl http://packs.download.atmel.com/Atmel.SAM3X_DFP.1.0.50.pack -L >Atmel.SAM3X_DFP.1.0.50.pack
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 1423k  100 1423k    0     0   415k      0  0:00:03  0:00:03 --:--:--  683k
```

flit, 2019-10-05:
I tried downloading Atmel.SAM3X _DFP.1.0.50.pack. CPM didn't report any errors, but the resulting file was this:
```html
<html>
<head><title>500 Internal Server Error</title></head>
<body>
<h1>500 Internal Server Error</h1>
<ul>
<li>Code: InternalError</li>
<li>Message: We encountered an internal error. Please try again.</li>
<li>RequestId: F4DDF32A05A6C7D9</li>
<li>HostId: ZRltk+iiSk0lknXexLow0IuV0S6P6Z15Y12ds+l8z/wOKcvZZGF1q/YWVQ0+b/MxwZZ6Uo+xk6E=</li>
</ul>
<hr/>
</body>
</html>
```

curl works fine for me with the SAM3X pack, too.

theotherjimmy, 2019-10-05:
Ah, dang. I'll check if `reqwest`'s async api implements redirects, @flit has found a redirect bug.

theotherjimmy, 2019-10-05:
It looks like ridirection support (https://docs.rs/reqwest/0.9.21/reqwest/async/struct.ClientBuilder.html#method.redirect) it's already built into `reqwest`. It also looks like native proxy support is there.

I'm going to try replacing `hyper` with the higher level `reqwest` in the coming days.

flit, 2019-10-06:
@theotherjimmy Thanks!

JanneKiiskila, 2019-10-07:
I can confirm that `reqwest` will succesfully download the files in question, I tried it out. 

```
extern crate reqwest;
extern crate tempfile;

use std::io::copy;
use std::fs::File;
//use tempfile::Builder;
use std::error::Error;

fn main() -> Result<(), Box<dyn Error>> {
    let target = "https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc";
    let max_redir = 3 as usize;
    reqwest::RedirectPolicy::limited(max_redir);
    let mut response = reqwest::get(target)?;
    println!("response = {:?}", response);

    let mut dest = {
        let fname = response
            .url()
            .path_segments()
            .and_then(|segments| segments.last())
            .and_then(|name| if name.is_empty() { None } else { Some(name) })
            .unwrap_or("tmp.bin");

        println!("file to download: '{}'", fname);
        let fname = "Keil.STM32F7xx_DFP.pdsc";
        println!("will be located under: '{:?}'", fname);
        File::create(fname)?
    };
    copy(&mut response, &mut dest)?;
    Ok(())
}
```

Since I have around 1 hour of experience with Rust, I didn't manage to do anything complex but enough to prove that it works.

```
jankii01@ubuntu:~/rust/downloader$ cargo run
   Compiling downloader v0.1.0 (/home/jankii01/rust/downloader)
    Finished dev [unoptimized + debuginfo] target(s) in 5.26s
     Running `target/debug/downloader`
response = Response { url: "http://sadevicepacksprodus.blob.core.windows.net/pdsc/Keil.STM32F7xx_DFP.pdsc", status: 200, headers: {"content-length": "343215", "content-type": "application/octet-stream", "content-md5": "AiE8GggHegSuHh096hBFEA==", "last-modified": "Wed, 17 Jul 2019 14:47:58 GMT", "etag": "0x8D70AC5C275FCCD", "server": "Windows-Azure-Blob/1.0 Microsoft-HTTPAPI/2.0", "x-ms-request-id": "37f29562-e01e-0064-32da-7c5da2000000", "x-ms-version": "2009-09-19", "x-ms-lease-status": "unlocked", "x-ms-blob-type": "BlockBlob", "date": "Mon, 07 Oct 2019 06:42:07 GMT"} }
file to download: 'Keil.STM32F7xx_DFP.pdsc'
will be located under: '"Keil.STM32F7xx_DFP.pdsc"'
```

And the file is downloaded correctly;

```
-rw-rw-r-- 1 jankii01 jankii01  343215 Oct  7 09:42 Keil.STM32F7xx_DFP.pdsc
```

theotherjimmy, 2019-10-07:
So much for @JanneKiiskila being a Rust noob. Thanks for confirming that switching to reqwest will fix this issue, Janne.

jeromecoutant, 2019-10-07:
Hi
Maybe using reqwest will solve #101 ? :-)

theotherjimmy, 2019-10-07:
@jeromecoutant, I think so, so long as we call https://docs.rs/reqwest/0.9.21/reqwest/async/struct.ClientBuilder.html#method.use_sys_proxy

theotherjimmy, 2019-10-07:
@JanneKiiskila I think we have further narrowed this down to a rustls-related issue, as the following diff makes it go away. This diff is applied to my https://github.com/theotherjimmy/cmsis-pack-manager/tree/use-reqwest branch.

```diff
❯ git diff
diff --git a/rust/cmsis-update/Cargo.toml b/rust/cmsis-update/Cargo.toml
index 91dd9b49b..c96bce7ad 100644
--- a/rust/cmsis-update/Cargo.toml
+++ b/rust/cmsis-update/Cargo.toml
@@ -19,5 +19,3 @@ pdsc = { path = "../pdsc" }

 [dependencies.reqwest]
 version = "0.9.21"
-default_features = false
-features = ["rustls-tls"]
diff --git a/rust/cmsis-update/src/download.rs b/rust/cmsis-update/src/download.rs
index 612bbaf23..43c680b13 100644
--- a/rust/cmsis-update/src/download.rs
+++ b/rust/cmsis-update/src/download.rs
@@ -134,7 +134,6 @@ where
 {
     pub fn new(config: &'a Conf, prog: Prog, log: &'a Logger) -> Result<Self, Error> {
         let client = ClientBuilder::new()
-            .use_rustls_tls()
             .use_sys_proxy()
             .redirect(RedirectPolicy::limited(5))
             .build()?;
                                 
```

kabirz, 2019-11-04:
always failed:
$ pyocd pack -u
Nov 04 14:24:26.168 ERRO download of "https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc: error trying to connect: Connection reset by peer (os error 104)
Nov 04 14:24:26.192 ERRO download of "https://www.keil.com/pack/Keil.STM32G0xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32G0xx_DFP.pdsc: error trying to connect: Connection reset by peer (os error 104)
Nov 04 14:24:26.307 ERRO download of "https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc: error trying to connect: Connection reset by peer (os error 104)
Nov 04 14:24:26.617 ERRO download of "http://www.qorvo.com/extra/keil_pack/Active-Semi.PAC52XX.pdsc" failed: https://www.qorvo.com/extra/keil_pack/Active-Semi.PAC52XX.pdsc: error trying to connect: Connection reset by peer (os error 104)
Nov 04 14:24:26.633 ERRO download of "http://www.qorvo.com/extra/keil_pack/Active-Semi.PAC55XX.pdsc" failed: https://www.qorvo.com/extra/keil_pack/Active-Semi.PAC55XX.pdsc: error trying to connect: Connection reset by peer (os error 104)
Downloading descriptors (518/540)

JanneKiiskila, 2019-11-04:
That is unfortunately the expected results now when using the RusTLS. It only works without it, but if we do it without it - we can't do a fully generic PIP wheel binary (which PIP requires).

theotherjimmy, 2019-11-08:
Dang, the number of failures increased to 5. That's not great.

JanneKiiskila, 2019-12-17:
Number of failures has grown even more, seems likely almost every NXP pack fails now. So, the problems keeps growing.
@madchutney  @theotherjimmy @mark-edgeworth  @bulislaw

erwango, 2020-01-13:
Any update on this point ?

rapgenic, 2020-11-22:
Is there any news on this? 

As of now pyOCD gives me the following errors when updating the index:

```
Nov 22 10:47:14.496 ERRO download of "https://www.keil.com/pack/Keil.STM32L4xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32L4xx_DFP.pdsc: error trying to connect: unexpected end of file
Nov 22 10:47:15.168 ERRO download of "https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc: error trying to connect: unexpected end of file
Nov 22 10:47:15.819 ERRO download of "https://www.keil.com/pack/Keil.STM32L5xx_DFP.pdsc" failed: https://www.keil.com/pack/Keil.STM32L5xx_DFP.pdsc: error trying to connect: unexpected end of file
```

Furthermore, when I look at `~/.local/share/cmsis-pack-manager`, a few pdsc files are _definitely_ not pdsc inside (but contain strange html/js code, I suppose as a result of a missed redirection/missing resource). The list of those files is:
* ABOV.CM0_DFP.1.0.4.pdsc
* ABOV.CM3_DFP.1.2.5.pdsc
* MindMotion.MM32x031_DFP.1.0.0.pdsc
* MindMotion.MM32x103_DFP.1.1.0.pdsc
* Redpine.RS13100_DFP.1.0.0.pdsc
* Redpine.RS14100_DFP.1.0.6.pdsc
* SiliconLabs.ARTIKBG1_DFP.5.7.0.pdsc
* SiliconLabs.ARTIKMG1_DFP.5.7.0.pdsc
* Sinowealth.SH32Fxxx_DFP.1.0.0.pdsc

flit, 2020-11-23:
@rapgenic Apologies, it keeps being fixed and then broken again. _\*sigh\*_ 

Thanks for reporting these problems. I'm putting in a request for this latest case to be fixed.

The root cause is that the keil.com server is old. Its TLS configuration doesn't include any modern ciphers that are considered secure, so the Rust rustls crate refuses to connect (cmsis-pack-manager's backend is implemented in Rust for performance). Sometimes when the list of Keil packs is updated, the author uses an https scheme for the pack, and thus cmsis-pack-manager via rustls will not allow it to be accessed.

[TLS analysis report for keil.com](https://www.ssllabs.com/ssltest/analyze.html?d=www.keil.com)

The second issue may be caused by errors by the silicon vendors, but clearly needs investigation. (And CPM should be detecting that the files are not XML, anyway.) I created an issue in the [cmsis-pack-manager](https://github.com/pyocd/cmsis-pack-manager) project.

jkrech, 2020-11-23:
I thought the keil web site was supposed to be "fixed" by now?!

flit, 2020-11-24:
@rapgenic It's fixed now!

rapgenic, 2020-11-24:
@flit Thank you very much!

str4t0m, 2021-05-01:
Since a while cmsis-packs for several microcontrollers can't be downloaded in pyocd.
I do assume this is still related to this problem.
The warning printed from pyocd is:
`0000628:WARNING:__main__:No matching devices. Please make sure the pack index is up to date.`

A few examples where the problem currently exists.

- ST STM32G0: Keil.STM32G0xx_DFP.1.3.0.pack (G0 had once issure, now they have returned).
- Redpine RS1410: Redpine.RS14100_DFP.1.0.6.pack (also RS1310).
- Microchip SAMD51: Microchip.SAMD51_DFP.3.4.91.pack (also several other samd, saml, and samr packages).
- Cypress PSOC 4 and PSOC6
- AmbiqMicro: AmbiqMicro.Apollo_DFP.1.2.0.pack
-  Renesas: Renesas.RA_DFP.3.0.0.pack

At least the STM32G0xx pack can be used in pyocd by manually downloading it and supplying it's path as argument.
Could this still be a problem with the website, or are these all related to #154 (invalid PDSC downloads).

JanneKiiskila, 2022-03-25:
I see something has been done now; 

https://github.com/pyocd/cmsis-pack-manager/commit/6ea727fad8af246b5384a00ee0ccbfbc9a571b6f

Has the issue been fixed/resolved now? @flit @mbrossard ?