pyocd/cmsis-pack-manager #122: Relax PDSC parsing rules

flit, 2019-10-05:
The `NordicSemiconductor.nRF_DeviceFamilyPack.8.24.1` PDSC causes CPM to spew 12 of these errors:
```
Oct 05 11:32:49.098 ERRO when trying to parse component: child element "description" not found in "component" element, Package: nRF_DeviceFamilyPack, Vendor: NordicSemiconductor
```

It seems like `description` elements should not cause an error if missing, even though they are technically required by the schema. At least this should be changed to a warning, or better, to a debug message. (I know it doesn't stop the PDSC from being parsed, but it might unduly worry users.)

theotherjimmy, 2019-10-06:
Totally agreed. It should be possible to change a `String` to an `Option<String>` and the related code for a single struct to acomplish this.

flit, 2019-10-06:
Nice, thanks!

jkrech, 2019-10-07:
Hi Chris,

The Keil.com pack importer refuses to update the version of the pack due to these Schema Check Errors:

NordicSemiconductor pack error(s):
Pack
Error(s)
NordicSemiconductor.nRF_DeviceFamilyPack.pdsc<http://developer.nordicsemi.com/nRF5_SDK/pieces/nRF_DeviceFamilyPack/NordicSemiconductor.nRF_DeviceFamilyPack.pdsc>
Error: The 'http://www.w3.org/2001/XMLSchema:noNamespaceSchemaLocation' attribute is not declared. Line Number: 2 Line Position: 74
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 756 Line Position: 6
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 761 Line Position: 6
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 771 Line Position: 6
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 781 Line Position: 6
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 791 Line Position: 6
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 801 Line Position: 6
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 811 Line Position: 6
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 821 Line Position: 6
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 831 Line Position: 6
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 841 Line Position: 6
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 851 Line Position: 6
Error: The element 'component' has incomplete content. List of possible elements expected: 'description'. Line Number: 861 Line Position: 6

This is the reason why we are still reporting version 8.24.1 as the latest available pack version.
We have already contacted Nordic a couple of times, however without getting a response.

Kind regards,

Joachim

Joachim Krech  |  Phone: +49 89 2620369-40 |  Email: joachim.krech@arm.com<mailto:joachim.krech@arm.com> |  www.keil.com<http://www.keil.com/>
ARM Germany GmbH  |  Bretonischer Ring 16  |  D-85630 Grasbrunn,
Germany Sitz der Gesellschaft: Grasbrunn  |  Handelsregister: München (HRB 175362)
Geschäftsführer: Andrew Smith, Joachim Krech, Reinhard Keil

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.