pyocd/cmsis-pack-manager #126: STM32F207ZG sectors not written out (sectors : null)

JanneKiiskila, 2019-10-18:
See originating PR; https://github.com/ARMmbed/mbed-os/pull/11707

The Keil CMSIS-pack file has this for the

```
        <!-- *************************  Device 'STM32F207ZG'  ***************************** -->
        <device Dname="STM32F207ZG">
          <memory id="IROM1"                               start="0x08000000" size="0x00100000" startup="1" default="1"/>
          <memory id="IRAM1"                               start="0x20000000" size="0x00020000" init   ="0" default="1"/>
          <algorithm name="CMSIS/Flash/STM32F2xx_1024.FLM" start="0x08000000" size="0x00100000"             default="1"/>
          <algorithm name="CMSIS/Flash/STM32F2xx_OPT.FLM"  start="0x1FFFC000" size="0x00000010"             default="0"/>
          <algorithm name="CMSIS/Flash/STM32F2xx_OTP.FLM"  start="0x1FFF7800" size="0x00000210"             default="0"/>
          <feature type="ADC"           n="24"      m="12"/>
          <feature type="IOs"           n="114"/>
          <variant Dvariant="STM32F207ZGTx"> <feature type="QFP" n="144"/> </variant>
        </device>
```

The current `index.json` under `tools/arm_pack_manager` has this information:

```
        "name": "STM32F207ZGTx", 
        "processor": {
            "Symmetric": {
                "core": "CortexM3", 
                "fpu": "None", 
                "mpu": "Present", 
                "units": 1
            }
        }, 
        "sectors": [
            [
                134217728, 
                16384
            ], 
            [
                134283264, 
                65536
            ], 
            [
                134348800, 
                131072
            ], 
            [
                536836096, 
                528
            ], 
            [
                536854528, 
                16
            ]
        ], 
```

So, the information seems to be essentially there, but it is not parsed out somehow / generated to the output.

theotherjimmy, 2019-10-18:
@JanneKiiskila This package does not populate sectors.

JanneKiiskila, 2019-10-29:
So, should the issue be moved over to Mbed OS?

theotherjimmy, 2019-10-29:
Yeah, that's a good idea.

JanneKiiskila, 2019-11-01:
Created new issue, cant' transfer due to sync script issues.
-> see https://github.com/ARMmbed/mbed-os/issues/11798

closed 2019-11-01 by JanneKiiskila