pyocd/cmsis-pack-manager #127: Release on crates.io

Yatekii, 2019-10-19:
Hi!

Any chance this is making it to crates.io with proper docs soon?

Best,
Yatekii

theotherjimmy, 2019-10-19:
Sure! What crates do you need published?

Yatekii, 2019-10-19:
I think the correct answer here for me is all of them :P

I am writing https://github.com/probe-rs/probe-rs and I want to extract the flash blob from the CMSIS pack. Maybe you could quickly elaborate if that's even possible.
Because in all honesty I didn't find much code which does this and I didn't fully see through what this collection of crates actually does =)

theotherjimmy, 2019-10-21:
@Yatekii That's very possible. I think pyOCD can do it with the python wrapper of these packages. I'll release the 0.1.0 with modified crate names. Checking for collisions right now.

Yatekii, 2019-10-21:
Great! Looking forward to it! If you could point me to the relevant bits once its out, that would be great :)

theotherjimmy, 2019-10-21:
Yeah, I'll post links in this issue as I publish the crates.

Yatekii, 2019-10-29:
Sorry to bug, but do you have an ETA on this? :)

theotherjimmy, 2019-10-29:
I don't. Thanks for reminding me to do this

mathias-arm, 2020-06-24:
Crate is now published at https://crates.io/crates/cmsis-pack let me know if there is an issue.

closed 2020-07-14 by mathias-arm