pyocd/cmsis-pack-manager #130: Unifying the crates

Yatekii, 2019-10-31:
Hi!

I just had a closer look at this lib. Is there a specific you made 8 small crates instead of a big one?
I think it's highly preferrable to have one single crate since all of this functionality is tightly coupled.

What do you think about this?

Best,
Yatekii

theotherjimmy, 2019-11-04:
@Yatekii I started this project before incremental compile was enabled on nightly. I'd be happy to unify most of the crates, excluding cmsis-cffi and cmsis-cli, as I don't expect users of the Rust crates to care about a foreign function interface, or an incomplete CLI.

Yatekii, 2019-11-04:
Sounds great! Should I make a PR or do you do it yourself?

theotherjimmy, 2019-11-05:
@Yatekii I don't currently have much time allocated to cmsis-pack-manager. I would appreciate a PR if you have the time.

PR #141 (2020-06-22): Merge crates (pack-index, cmsis-update, pdsc, utils)
Tentative PR for Unifying the crates

Fixes #130
closed 2020-06-22 by mathias-arm