pyocd/cmsis-pack-manager #137: Install with pip >=20 always triggers build

juhhov, 2020-01-23:
Downgrading to < 20 uses wheel.
```
$ pip --version
pip 20.0.1 from /home/juho/py3/lib/python3.5/site-packages/pip (python 3.5)
(py3) juho@ub:~$ pip install cmsis-pack-manager
Collecting cmsis-pack-manager
  Using cached cmsis-pack-manager-0.2.9.tar.gz (42 kB)
Collecting appdirs>=1.4
  Using cached appdirs-1.4.3-py2.py3-none-any.whl (12 kB)
Collecting milksnake>=0.1.2
  Using cached milksnake-0.1.5-py2.py3-none-any.whl (9.6 kB)
Processing ./.cache/pip/wheels/e4/76/4d/a95b8dd7b452b69e8ed4f68b69e1b55e12c9c9624dd962b191/PyYAML-5.3-cp35-cp35m-linux_x86_64.whl
Collecting cffi>=1.6.0
  Using cached cffi-1.13.2-cp35-cp35m-manylinux1_x86_64.whl (397 kB)
Processing ./.cache/pip/wheels/f2/9a/90/de94f8556265ddc9d9c8b271b0f63e57b26fb1d67a45564511/pycparser-2.19-py2.py3-none-any.whl
Building wheels for collected packages: cmsis-pack-manager
  Building wheel for cmsis-pack-manager (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/juho/py3/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-aa67zdj0/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/tmp/pip-install-aa67zdj0/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-tar7vy6_
       cwd: /tmp/pip-install-aa67zdj0/cmsis-pack-manager/
  Complete output (9 lines):
  running bdist_wheel
  running build
  running build_py
  creating build/lib
  creating build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/__init__.py -> build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/_version.py -> build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/pack_manager.py -> build/lib/cmsis_pack_manager
  error: [Errno 2] No such file or directory: 'cargo'
  ----------------------------------------
  ERROR: Failed building wheel for cmsis-pack-manager
  Running setup.py clean for cmsis-pack-manager
Failed to build cmsis-pack-manager
Installing collected packages: appdirs, pycparser, cffi, milksnake, pyyaml, cmsis-pack-manager
    Running setup.py install for cmsis-pack-manager ... error
    ERROR: Command errored out with exit status 1:
     command: /home/juho/py3/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-aa67zdj0/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/tmp/pip-install-aa67zdj0/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-jt_ysbz4/install-record.txt --single-version-externally-managed --compile --install-headers /home/juho/py3/include/site/python3.5/cmsis-pack-manager
         cwd: /tmp/pip-install-aa67zdj0/cmsis-pack-manager/
    Complete output (9 lines):
    running install
    running build
    running build_py
    creating build/lib
    creating build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/__init__.py -> build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/_version.py -> build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/pack_manager.py -> build/lib/cmsis_pack_manager
    error: [Errno 2] No such file or directory: 'cargo'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/juho/py3/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-aa67zdj0/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/tmp/pip-install-aa67zdj0/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-jt_ysbz4/install-record.txt --single-version-externally-managed --compile --install-headers /home/juho/py3/include/site/python3.5/cmsis-pack-manager Check the logs for full command output.
(py3) juho@ub:~$ pip install pip==19.3.1
Collecting pip==19.3.1
  Using cached pip-19.3.1-py2.py3-none-any.whl (1.4 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.0.1
    Uninstalling pip-20.0.1:
      Successfully uninstalled pip-20.0.1
Successfully installed pip-19.3.1
(py3) juho@ub:~$ pip install cmsis-pack-manager
Collecting cmsis-pack-manager
  Using cached https://files.pythonhosted.org/packages/fd/6b/b1f8f4117b3a4efafbdb04eaa7e7685a395cda565b3b221860d04f2a827c/cmsis_pack_manager-0.2.9-py2.py3-none-manylinux1_x86_64.whl
Requirement already satisfied: pyyaml>=3.12 in ./py3/lib/python3.5/site-packages (from cmsis-pack-manager) (5.3)
Requirement already satisfied: milksnake>=0.1.2 in ./py3/lib/python3.5/site-packages (from cmsis-pack-manager) (0.1.5)
Requirement already satisfied: appdirs>=1.4 in ./py3/lib/python3.5/site-packages (from cmsis-pack-manager) (1.4.3)
Requirement already satisfied: cffi>=1.6.0 in ./py3/lib/python3.5/site-packages (from milksnake>=0.1.2->cmsis-pack-manager) (1.13.2)
Requirement already satisfied: pycparser in ./py3/lib/python3.5/site-packages (from cffi>=1.6.0->milksnake>=0.1.2->cmsis-pack-manager) (2.19)
Installing collected packages: cmsis-pack-manager
Successfully installed cmsis-pack-manager-0.2.9
WARNING: You are using pip version 19.3.1; however, version 20.0 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

acabarbaye, 2020-01-23:
@juhhov Thanks for finding this out. We experience the same issue.
The implications of this problem are much wider than just the installation of this package as it makes `mbed import` command to fail if `pip` has been upgraded.
We will try to find a remedy ASAP.

acabarbaye, 2020-01-23:
This may actually be caused by this issue: https://github.com/pypa/pip/issues/7629

juhhov, 2020-01-29:
Pip 20.0.2 fixes this.

closed 2020-01-29 by juhhov

marcel-bluestone, 2021-04-16:
I still do experience an error with pip 20.3.4

Collecting cmsis-pack-manager<0.3.0,>=0.2.3
  Using cached cmsis-pack-manager-0.2.10.tar.gz (45 kB)
    ERROR: Command errored out with exit status 1:
     command: /home/marcel/.pyenv/versions/2.7.18/bin/python2.7 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-t6pg_T/cmsis-pack-manager/setup.py'"'"'; __file__='"'"'/tmp/pip-install-t6pg_T/cmsis-pack-manager/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-y36jvN
         cwd: /tmp/pip-install-t6pg_T/cmsis-pack-manager/
    Complete output (60 lines):
    /tmp/easy_install-TGE31H/setuptools_scm-6.0.1/src
    <pkg_resources.WorkingSet object at 0xffffa6fc9250>
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-t6pg_T/cmsis-pack-manager/setup.py", line 73, in <module>
        test_suite="tests"
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/__init__.py", line 144, in setup
        _install_setup_requires(attrs)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/dist.py", line 719, in fetch_build_eggs
        replace_conflicting=True,
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/pkg_resources/__init__.py", line 782, in resolve
        replace_conflicting=replace_conflicting
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1065, in best_match
        return self.obtain(req, installer)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1077, in obtain
        return installer(requirement)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/dist.py", line 786, in fetch_build_egg
        return cmd.easy_install(req)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 679, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 705, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 890, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1158, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1144, in run_setup
        run_setup(setup_script, args)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/sandbox.py", line 253, in run_setup
        raise
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/sandbox.py", line 166, in save_modules
        saved_exc.resume()
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/sandbox.py", line 141, in resume
        six.reraise(type, exc, self._tb)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/sandbox.py", line 154, in save_modules
        yield saved
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/sandbox.py", line 250, in run_setup
        _execfile(setup_script, ns)
      File "/home/marcel/.pyenv/versions/2.7.18/lib/python2.7/site-packages/setuptools/sandbox.py", line 45, in _execfile
        exec(code, globals, locals)
      File "/tmp/easy_install-TGE31H/setuptools_scm-6.0.1/setup.py", line 52, in <module>
        with open("test_requirements.txt") as testreq:
      File "/tmp/easy_install-TGE31H/setuptools_scm-6.0.1/setup.py", line 29, in scm_config
        spec.add_cffi_module(
      File "/tmp/easy_install-TGE31H/setuptools_scm-6.0.1/src/setuptools_scm/__init__.py", line 8, in <module>
      File "/tmp/easy_install-TGE31H/setuptools_scm-6.0.1/src/setuptools_scm/config.py", line 6, in <module>
      File "/tmp/easy_install-TGE31H/setuptools_scm-6.0.1/src/setuptools_scm/utils.py", line 41
        print(*k)
              ^
    SyntaxError: invalid syntax
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
`
`

marcel-bluestone, 2021-04-16:
downgrading to pip 20.0.2 did not help unfortunately

flit, 2021-09-14:
@marcel-bluestone It looks like you are using Python 2, which is not supported by cmsis-pack-manager any more. (I'm surprised pip even tried to install that version of setuptools_scm, which also does not support Python 2.)

zacpod, 2021-10-14:
@flit Thanks - I am on Pip 21.3, but still wasn't working.  
But installing Rust worked a treat.  
Appreciate the quick response.  :)

flit, 2021-10-14:
Ok, thanks for the update! Not sure what the problem is then, we'll have to investigate. I'll create a new issue to track it, since it seems to be a different cause.

zacpod, 2021-10-14:
@flit Let me know if I can help.  I'm in the process of getting my Nucleo-H743ZI2 up and running with mbed, and so far it's been fairly easy - but pyocd doesn't work yet in Window or Linux.  I'm sure it's mostly pebcak as I'm very new to all this, but if I can help test or shoot logs over to you I'll be more than happy to do so.

flit, 2021-10-14:
Fyi, the issue with Python 3.10 is simple: we didn't release wheels for Python 3.10 since it wasn't final. Working on getting a wheel out. 😄