pyocd/cmsis-pack-manager #138: Custom pack URLs

flit, 2020-02-29:
Microchip publishes packs at https://packs.download.microchip.com/ It would be awesome to support these packs, or other custom urls.

Copied from mbedmicro/pyOCD#823.

theotherjimmy, 2020-03-16:
You should already be able to do that by adding the pack index url into a config file. I forget where it's stored, something ending with `.list` IIRC.

s-celles, 2025-07-14:
Hi,

Currently working on https://github.com/platformio/platformio-core/issues/704#issuecomment-3041247240
Providing an easy way to download DFP from that site will be great through your Python lib could be great.
About pack index url... it's ending with `/Microchip.pidx`
Kind regards