pyocd/cmsis-pack-manager #139: Installation error

Sup3rGeo, 2020-03-16:
Hi,

I am trying to install this package in my raspberry pi on linux (raspbian):

Getting a stable version of rust:

```
(venv) pi@raspberrypi:~ $ rustup update stable
info: syncing channel updates for 'stable-armv7-unknown-linux-gnueabihf'

  stable-armv7-unknown-linux-gnueabihf unchanged - rustc 1.42.0 (b8cedc004 2020-03-09)

info: checking for self-updates
```

Installing from master:

```
(venv) pi@raspberrypi:~ $ pip install git+https://github.com/ARMmbed/cmsis-pack-manager.git
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/ARMmbed/cmsis-pack-manager.git
  Cloning https://github.com/ARMmbed/cmsis-pack-manager.git to /tmp/pip-req-build-90tutd2x
Requirement already satisfied: appdirs>=1.4 in ./race-automated-testing/venv/lib/python3.7/site-packages (from cmsis-pack-manager==0.2.10.dev4) (1.4.3)
Requirement already satisfied: milksnake>=0.1.2 in ./race-automated-testing/venv/lib/python3.7/site-packages (from cmsis-pack-manager==0.2.10.dev4) (0.1.5)
Requirement already satisfied: pyyaml>=3.12 in ./race-automated-testing/venv/lib/python3.7/site-packages (from cmsis-pack-manager==0.2.10.dev4) (5.3)
Requirement already satisfied: cffi>=1.6.0 in ./race-automated-testing/venv/lib/python3.7/site-packages (from milksnake>=0.1.2->cmsis-pack-manager==0.2.10.dev4) (1.14.0)
Requirement already satisfied: pycparser in ./race-automated-testing/venv/lib/python3.7/site-packages (from cffi>=1.6.0->milksnake>=0.1.2->cmsis-pack-manager==0.2.10.dev4) (2.20)
Building wheels for collected packages: cmsis-pack-manager
  Running setup.py bdist_wheel for cmsis-pack-manager ... error
  Complete output from command /home/pi/race-automated-testing/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-90tutd2x/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-yrzxacot --python-tag cp37:
  running bdist_wheel
  running build
  running build_py
  creating build/lib
  creating build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/__init__.py -> build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/pack_manager.py -> build/lib/cmsis_pack_manager
  copying cmsis_pack_manager/_version.py -> build/lib/cmsis_pack_manager
      Updating crates.io index
     Compiling libc v0.2.67
     Compiling cfg-if v0.1.10
     Compiling autocfg v1.0.0
     Compiling lazy_static v1.4.0
     Compiling semver-parser v0.7.0
     Compiling byteorder v1.3.4
     Compiling futures v0.1.29
     Compiling cc v1.0.50
     Compiling log v0.4.8
     Compiling either v1.5.3
     Compiling maybe-uninit v2.0.0
     Compiling slab v0.4.2
     Compiling scopeguard v1.1.0
     Compiling proc-macro2 v1.0.9
     Compiling unicode-xid v0.2.0
     Compiling syn v1.0.16
     Compiling fnv v1.0.6
     Compiling version_check v0.9.1
     Compiling rand_core v0.4.2
     Compiling autocfg v0.1.7
     Compiling matches v0.1.8
     Compiling smallvec v1.2.0
     Compiling itoa v0.4.5
     Compiling rustc-demangle v0.1.13
     Compiling spin v0.5.2
     Compiling untrusted v0.7.0
     Compiling failure_derive v0.1.7
     Compiling slog v2.5.2
     Compiling serde v1.0.104
     Compiling ryu v1.0.3
     Compiling percent-encoding v1.0.1
     Compiling httparse v1.3.4
     Compiling try-lock v0.2.2
     Compiling regex-syntax v0.6.17
     Compiling percent-encoding v2.1.0
     Compiling crc32fast v1.2.0
     Compiling adler32 v1.0.4
     Compiling encoding_rs v0.8.22
     Compiling mime v0.3.16
     Compiling dtoa v0.4.5
     Compiling take_mut v0.2.2
     Compiling scoped-tls v0.1.2
     Compiling encoding_rs v0.6.11
     Compiling try_from v0.3.2
  error[E0214]: parenthesized type parameters may only be used with a `Fn` trait
    --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/encoding_rs-0.6.11/src/utf_16.rs:33:44
     |
  33 |     pub fn additional_from_state(&self) -> usize() {
     |                                            ^^^^^^^ only `Fn` traits may use parentheses
  
     Compiling thread_local v1.0.1
  error: aborting due to previous error
  
  For more information about this error, try `rustc --explain E0214`.
  error: could not compile `encoding_rs`.
  warning: build failed, waiting for other jobs to finish...
  error: build failed
  
  ----------------------------------------
  Failed building wheel for cmsis-pack-manager
  Running setup.py clean for cmsis-pack-manager
Failed to build cmsis-pack-manager
Installing collected packages: cmsis-pack-manager
  Running setup.py install for cmsis-pack-manager ... error
    Complete output from command /home/pi/race-automated-testing/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-90tutd2x/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-8gzoqi08/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/race-automated-testing/venv/include/site/python3.7/cmsis-pack-manager:
    running install
    running build
    running build_py
    creating build/lib
    creating build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/__init__.py -> build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/pack_manager.py -> build/lib/cmsis_pack_manager
    copying cmsis_pack_manager/_version.py -> build/lib/cmsis_pack_manager
       Compiling libc v0.2.67
       Compiling semver v0.9.0
       Compiling crossbeam-utils v0.7.2
       Compiling crossbeam-epoch v0.8.2
       Compiling indexmap v1.3.2
       Compiling num-traits v0.2.11
       Compiling encoding_rs v0.6.11
       Compiling num-integer v0.1.42
    error[E0214]: parenthesized type parameters may only be used with a `Fn` trait
      --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/encoding_rs-0.6.11/src/utf_16.rs:33:44
       |
    33 |     pub fn additional_from_state(&self) -> usize() {
       |                                            ^^^^^^^ only `Fn` traits may use parentheses
    
       Compiling crossbeam-utils v0.6.6
       Compiling byteorder v1.3.4
       Compiling log v0.4.8
       Compiling backtrace-sys v0.1.34
    error: aborting due to previous error
    
    For more information about this error, try `rustc --explain E0214`.
    error: could not compile `encoding_rs`.
    warning: build failed, waiting for other jobs to finish...
    error: build failed
    
    ----------------------------------------
Command "/home/pi/race-automated-testing/venv/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-90tutd2x/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-8gzoqi08/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/race-automated-testing/venv/include/site/python3.7/cmsis-pack-manager" failed with error code 101 in /tmp/pip-req-build-90tutd2x/
```

This (https://github.com/probe-rs/target-gen/issues/9) suggests that it should compile with `rustc` version `1.40.0` but when I try installing it I get:

```
(venv) pi@raspberrypi:~ $ rustup install 1.40.0
info: syncing channel updates for '1.40.0-armv7-unknown-linux-gnueabihf'
info: latest update on 2019-12-19, rust version 1.40.0 (73528e339 2019-12-16)
error: component 'clippy' for target 'armv7-unknown-linux-gnueabihf' is unavailable for download for channel 1.40.0
```

I would appreciate any help. I am quite a noob with rust and this package, it is a dependency of another package I use in my tests.

Thanks!

Sup3rGeo, 2020-03-17:
Solved by manually changing dependencies and building it again.

closed 2020-03-17 by Sup3rGeo

theotherjimmy, 2020-03-17:
@Sup3rGeo For future people that come from a search, which dependencies did you change?

Sup3rGeo, 2020-03-23:
I updated

```
	modified:   rust/cmsis-update/Cargo.toml
	modified:   rust/pack-index/Cargo.toml
	modified:   rust/pdsc/Cargo.toml
	modified:   rust/utils/Cargo.toml
```

With 

```
minidom = “0.6.0”
Quick-xml = “0.9.0”
```