pyocd/cmsis-pack-manager #151: have aarch64 whl files

galak, 2020-09-10:
To make life easier would be nice if there was linux aarch64 whl on release.

theotherjimmy, 2020-09-10:
Has a standard come out for aarch64 wheels? I don't recall their being one when I last checked, 2 years ago.

flit, 2020-09-29:
AArch64 is supported now by `manylinux2014` defined by [PEP 599](https://www.python.org/dev/peps/pep-0599/).

See this blog:
https://marcin.juszkiewicz.com.pl/2019/12/04/python-and-aarch64/

galak, 2020-12-10:
Any updates on this?

mathias-arm, 2020-12-10:
We will try to add this when we make the next release.

pfalcon, 2021-02-08:
Note that cmsis-pack-manager is (again) the hard dependency of pyocd, in particular instructions describing how to get around need to install cmsis-pack-manager to use pyocd: https://github.com/pyocd/pyOCD/blob/master/docs/installing_on_non_x86.md - no longer work with the latest pyocd (0.29) and/or recent Python: https://github.com/pyocd/pyOCD/issues/1023 .

@galak, FYI

flit, 2021-09-13:
@galak cmsis-pack-manager 0.3.0 is released with AArch64 wheels. GitHub Actions workflows are set up so all future releases will support AArch64.

closed 2021-09-13 by flit