pyocd/cmsis-pack-manager #154: Investigate invalid PDSC downloads

flit, 2020-11-23:
(Reported by @rapgenic in pyocd/pyocd#121.)

In the cmsis-pack-manager local PDSC repository, a few .pdsc files are definitely not PDSC XML. Instead they contain strange html/js code. Is this a result of a missed redirection/missing resource, or is it simply invalid configuration on the part of the silicon vendors?

List of questionable files:

- ABOV.CM0_DFP.1.0.4.pdsc
- ABOV.CM3_DFP.1.2.5.pdsc
- MindMotion.MM32x031_DFP.1.0.0.pdsc
- MindMotion.MM32x103_DFP.1.1.0.pdsc
- Redpine.RS13100_DFP.1.0.0.pdsc
- Redpine.RS14100_DFP.1.0.6.pdsc
- SiliconLabs.ARTIKBG1_DFP.5.7.0.pdsc
- SiliconLabs.ARTIKMG1_DFP.5.7.0.pdsc
- Sinowealth.SH32Fxxx_DFP.1.0.0.pdsc

rapgenic, 2020-11-23:
Also, when updating the index, a few warnings are reported (which I originally ignored, being them warnings), like (note that there are many others and a lot of them are related to functioning packs which probably contain little errors):

```
Nov 23 17:13:27.339 WARN tried to parse element "package" from element "html"
Nov 23 17:13:27.454 WARN the XML is invalid, an element was wrongly closed
Nov 23 17:13:27.464 WARN Only Comment, CDATA and DOCTYPE nodes can start with a '!'
```

Which suggest that the library is recognising those files as problematic.

flit, 2021-08-30:
Most of the listed PDSCs are actually returning 404 Not Found responses. Because of #162 and #155, the html content of the error message sent by the server gets written to the .pdsc file.