pyocd/cmsis-pack-manager #157: Integrate with other pack repositories

flit, 2021-01-07:
Support integration with pack repositories from other tools:
- Keil µVision
- Arm CMSIS-Pack Eclipse plugin
- IAR
- Eclipse Embedded CDT?

Specifically, enable the features requested in this issue: ARM-software/CMSIS_5#1089.

There may be issues that prevent easy integration, but they should be identified.

fred-r, 2021-01-14:
Local repository support:
----------------------------

The idea is to be able to say that a pack points to a repository which is not the "CMSIS_PACK_ROOT" repository.
The folder we indciate would contain:
- the pdsc descrivbing the pack
- the artefacts composing the pack directly available

So, instead of installing the pack from an archive to CMSIS_PACK_ROOT, I want the pack to point to my local development folder where I can do changes and test them without generating and re-installing a pack each time.

Thanks & Regards,
Fred

jkrech, 2021-01-15:
The way this works today is that there is a package index file in .Local pointing to the location of a pdsc file at the root point of an expanded pack.

`$CMSIS_PACK_ROOT/.Local/local_repository.pidx`

local_repository.pidx:

```
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<index schemaVersion="1.1.0" xmlns:xs="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xs:noNamespaceSchemaLocation="PackIndex.xsd">
  <vendor>local repository</vendor>
  <url>file://localhost/C:/Users/anyuser/AppData/Local/Arm/Packs</url>
  <timestamp>2021-01-15T10:06:21</timestamp>
  <pindex>
    <pdsc name="MyTestPack" url="file://localhost/C:/mypacks/mytestpack/" vendor="MyVendor" version="0.0.1"/>
  </pindex>
</index>
```

fred-r, 2021-01-15:
Hi,

indeed I use to update this file when I do not want to use the Pack Installer UI and it is working fine.

Thans & Regards,
Fred