pyocd/cmsis-pack-manager #159: error 404 whentrying to access the docs

fred-r, 2021-01-13:
Hi,

the link given int he readme:
   https://armmbed.github.io/cmsis-pack-manager/
seems to be broken:
  error 404 returned.

Thanks & Regards,
Fred

fred-r, 2021-01-13:
Follow-up:

I cloned the cmsis-pack-manager github and did:
pip install cmsis-pack-manager

Then I tried:
python pack_manager.py

But this does not display any help and the documentation link seems to be broken.

Is it the proper way of using the pack manager ? (I see there are commands and subcommands but at the moment I do not know how to launch it, even the help text is not displayed).

Probably my way of calling the tool is very wrong ?

Thanks & Regards,
Fred

mbrossard, 2021-01-13:
The documentation should be available at this URL: https://pyocd.github.io/cmsis-pack-manager/

The `README.md` suggests using `python setup.py bdist_wheel` to build the package. The `build-wheels.sh` script should be helpful on how to build and install this way for Mac and Linux.

flit, 2021-01-13:
Hi @fred-r 

I've successfully recently built using `python setup.py bdist_wheel`.

The docs on the GitHub pages site are unfortunately very out of date; they are for the pre-Rust-based version of CPM. We should either regenerate and update the docs site or completely remove them.

fred-r, 2021-01-14:
Hi @flit,

the command you indicate does not work for me.

First, I installed the requirements:

$ pip install -r setup_requirements.txt
Requirement already satisfied: milksnake>=0.1.2 in c:\users\frq09468\appdata\local\programs\python\python37\lib\site-packages (from -r setup_requirements.txt (line 1)) (0.1.5)
Collecting pytest-runner
  Downloading pytest_runner-5.2-py2.py3-none-any.whl (6.8 kB)
Requirement already satisfied: setuptools>=40.0 in c:\users\frq09468\appdata\local\programs\python\python37\lib\site-packages (from -r setup_requirements.txt (line 3)) (51.1.2)
Collecting setuptools_scm!=1.5.3,!=1.5.4
  Downloading setuptools_scm-5.0.1-py2.py3-none-any.whl (28 kB)
Collecting setuptools_scm_git_archive
  Downloading setuptools_scm_git_archive-1.1-py2.py3-none-any.whl (3.3 kB)
Requirement already satisfied: cffi>=1.6.0 in c:\users\frq09468\appdata\local\programs\python\python37\lib\site-packages (from milksnake>=0.1.2->-r setup_requirements.txt (line 1)) (1.14.4)
Requirement already satisfied: pycparser in c:\users\frq09468\appdata\local\programs\python\python37\lib\site-packages (from cffi>=1.6.0->milksnake>=0.1.2->-r setup_requirements.txt (line 1)) (2.20)
Installing collected packages: pytest-runner, setuptools-scm, setuptools-scm-git-archive
Successfully installed pytest-runner-5.2 setuptools-scm-5.0.1 setuptools-scm-git-archive-1.1

Then I try the setup.py:

$ python setup.py bdist_wheel .
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'bdist_wheel'

As you can see the 'bdist_wheel' command is indicated as invalid.

Thanks & Regards,
Fred

flit, 2021-01-14:
You need to install wheel support with `pip install wheel`. (Sorry, it's often installed by default.)

fred-r, 2021-01-14:
Hi,

thanks for the hint.
It is better now but still failing:

$ python setup.py bdist_wheel
running bdist_wheel
running build
running build_py
copying cmsis_pack_manager\_version.py -> build\lib\cmsis_pack_manager
error: [WinError 2] The system cannot find the file specified

Thanks & Regards,
Fred

flit, 2021-01-14:
Do you have Rust installed (the CPM backend is built in Rust)? See https://rustup.rs. Sincere apologies if this wasn't clear in the Readme!

fred-r, 2021-01-14:
Ok, my bad, the readme is quite clear about it.
Sorry for any incovenience.

Thanks & Regards,
Fred

flit, 2021-01-24:
Closing this issue as the docs URL in the readme was fixed.

closed 2021-01-24 by flit