pyocd/cmsis-pack-manager #161: Question about local repositories support

fred-r, 2021-01-14:
Hi,

with the Keil uVision pack manager the packs are by default installed in the packs directory.
In my setup, this is:
C:\Keil_v5\ARM\PACK

But the pack installer also works with "local repositories".
I mean that I can indicate a folder containing a pdsc file and the required artefacts and this supersedes a similar pack definition in the root directory.

Does cmsis-pack-manager support the same "local repositories" feature ?
Or is it only working with the root directory and ".pack" files ?

Thanks & Regards,
Fred

flit, 2021-01-14:
At the moment, CPM only supports its own pack repository containing .pdsc files and .packs under subdirectories for the vendor and pack name. Later today I'll document structure this in a wiki page and reference it here.

CPM may very well not support your use case right now… But the goal is for it to be a common command-line tool for managing CMSIS-Packs. That's why I added #157. I'd like to document your use case in that issue.

fred-r, 2021-01-15:
Thanks for the clarification and the #157 issue.

closed 2021-01-15 by fred-r