pyocd/cmsis-pack-manager #187: Consider replacing milksnake with maturin

dvzrv, 2022-01-28:
Hi! It looks like [maturin](https://github.com/PyO3/maturin) is the more actively maintained project, whereas [milksnake](https://github.com/getsentry/milksnake) appears rather dead (no release in four years, open pull requests are ignored).

Would it be possible for this project to facilitate maturin instead of milksnake (unless I'm misinterpreting the scope).

I'm asking this because I'm packaging maturin on Arch Linux already and packaging a rather undead package for building cmsis-pack-manager could potentially be circumvented.

mathias-arm, 2022-01-28:
We have been considering it. When we moved to GitHub Actions and added support for `universal2` binaries on Mac, I tried using `maturin` but did not succeed. CFFI is what we use and that seems to be maturin's third priority (with PyO3 and rust-cpython ahead of CFFI) and there was not that many examples at the time. I am sure things have evolved since then.

If you (or someone else) can try to get `maturin` working with `cmsis-pack-manager` and `cibuidwheel` I would certainly give it another shot.

dvzrv, 2022-01-28:
@mathias-arm Thanks for getting back on this inquiry!

I'm afraid I lack the expertise (and time) to drive such a change for this project though. But maybe someone else has? :)

dvzrv, 2022-02-08:
FWIW, due to milksnake seemingly being abandonware it will stop working (as in, downstreams will not be able to package it without a lot of pain and will be more likely to remove it if that is the case) in the not too distant future, as it is not yet PEP517 ready (somewhat related PR to improve the setuptools integration: https://github.com/getsentry/milksnake/pull/30)

mbrossard, 2022-11-18:
Could you try to build the [maturin](https://github.com/pyocd/cmsis-pack-manager/tree/maturin) branch? I am not sure I have properly connected all the pieces. I need to update GitHub Actions configuration, and `setuptools_scm` needs to be replaced.

PR #195 (2022-12-17): Replace Milksnake by Maturin
- Replaces Milksnake by Maturin to build bindings (fixes #187, fixes #188, and fixes #194).
- Pins Rust version to 1.63.0 (version 1.64.0 raised GLIBC version requirements that are not compatible with the `manylinux` version used), this should fix building wheels in CI.
- Upgrade version of `cibuildwheel`. Since `cmsis-pack-manager` does not use Python API, the bindings works independently of the version of Python. This is detected by `cibuildwheel` since version [2.8.0](https://github.com/pypa/cibuildwheel/releases/tag/v2.8.0). Upgrading the version of This will r [... cut]
closed 2022-12-17 by flit