pyocd/cmsis-pack-manager #188: Is milksnake a runtime dependency?

dvzrv, 2022-02-08:
The [setup.cfg](https://github.com/pyocd/cmsis-pack-manager/blob/5b61d18925c1bbdc3a98c658b99c00b01690a44f/setup.cfg#L39) denotes milksnake as a runtime dependency, but I believe that is very likely an error (it's used during setup, but not during runtime).

It would be great to get confirmation on this, then I can open a PR to remove it :)

mbrossard, 2022-11-18:
Could you try to build the [maturin](https://github.com/pyocd/cmsis-pack-manager/tree/maturin) branch? I am not sure I have properly connected all the pieces. I need to update GitHub Actions configuration, and `setuptools_scm` needs to be replaced.

dvzrv, 2022-12-20:
milksnake is unfortunately still present in setup.cfg and will unconditionally drag it in upon installing this project.

PR #195 (2022-12-17): Replace Milksnake by Maturin
- Replaces Milksnake by Maturin to build bindings (fixes #187, fixes #188, and fixes #194).
- Pins Rust version to 1.63.0 (version 1.64.0 raised GLIBC version requirements that are not compatible with the `manylinux` version used), this should fix building wheels in CI.
- Upgrade version of `cibuildwheel`. Since `cmsis-pack-manager` does not use Python API, the bindings works independently of the version of Python. This is detected by `cibuildwheel` since version [2.8.0](https://github.com/pypa/cibuildwheel/releases/tag/v2.8.0). Upgrading the version of This will r [... cut]
PR #198 (2022-12-22): Multiple bug fixes
- Fix a vulnerability in `time` reported by dependabot.
- Remove (obsolete) runtime dependency on `milksnake` (fixes #188)
- Missing support for Python 3.11
commit 3f108f0b (2022-12-22): Remove milksnake runtime dependency

Fixes #188
closed 2022-12-22 by flit