pyocd/cmsis-pack-manager #194: Error Building Wheel with Python 3.11

banduccm, 2022-11-16:
I was attempting to update my Zephyr workspace after updating to Fedora 37 and it seems as though the wheel for cmsis-pack-manager no longer builds:

```
(.venv) [mbanducci zephyr]$ pip install cmsis-pack-manager
Collecting cmsis-pack-manager
  Using cached cmsis-pack-manager-0.4.0.tar.gz (43 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting appdirs<2.0,>=1.4
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting milksnake<1.0,>=0.1.5
  Using cached milksnake-0.1.5-py2.py3-none-any.whl (9.6 kB)
Requirement already satisfied: pyyaml<7.0,>=6.0 in /home/mbanducci/zephyrproject/.venv/lib64/python3.11/site-packages (from cmsis-pack-manager) (6.0)
Collecting cffi>=1.6.0
  Using cached cffi-1.15.1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (462 kB)
Collecting pycparser
  Using cached pycparser-2.21-py2.py3-none-any.whl (118 kB)
Building wheels for collected packages: cmsis-pack-manager
  Building wheel for cmsis-pack-manager (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for cmsis-pack-manager (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [285 lines of output]
      /tmp/pip-build-env-a67mmg56/overlay/lib/python3.11/site-packages/setuptools/config/setupcfg.py:508: SetuptoolsDeprecationWarning: The license_file parameter is deprecated, use license_files instead.
        warnings.warn(msg, warning_class)
      running bdist_wheel
      running build
      running build_py
      creating build/lib
      creating build/lib/cmsis_pack_manager
      copying cmsis_pack_manager/pack_manager.py -> build/lib/cmsis_pack_manager
      copying cmsis_pack_manager/_version.py -> build/lib/cmsis_pack_manager
      copying cmsis_pack_manager/_native__ffi.py -> build/lib/cmsis_pack_manager
      copying cmsis_pack_manager/_native.py -> build/lib/cmsis_pack_manager
      copying cmsis_pack_manager/__init__.py -> build/lib/cmsis_pack_manager
          Updating crates.io index
         Compiling libc v0.2.137
         Compiling autocfg v1.1.0
         Compiling cfg-if v1.0.0
         Compiling cfg-if v0.1.10
         Compiling futures v0.1.31
         Compiling log v0.4.17
         Compiling byteorder v1.4.3
         Compiling semver-parser v0.7.0
         Compiling either v1.8.0
         Compiling maybe-uninit v2.0.0
         Compiling lazy_static v1.4.0
         Compiling cc v1.0.76
         Compiling scopeguard v1.1.0
         Compiling proc-macro2 v1.0.47
         Compiling unicode-ident v1.0.5
         Compiling quote v1.0.21
         Compiling fnv v1.0.7
         Compiling tinyvec_macros v0.1.0
         Compiling syn v1.0.103
         Compiling rand_core v0.4.2
         Compiling unicode-bidi v0.3.8
         Compiling once_cell v1.16.0
         Compiling version_check v0.9.4
         Compiling spin v0.5.2
         Compiling untrusted v0.7.1
         Compiling tinyvec v1.6.0
         Compiling memchr v2.5.0
         Compiling lock_api v0.3.4
         Compiling matches v0.1.9
         Compiling serde_derive v1.0.147
         Compiling itoa v0.4.8
         Compiling serde v1.0.147
[... 222 lines cut ...]
        File "/usr/lib64/python3.11/re/__init__.py", line 227, in compile
          return _compile(pattern, flags)
                 ^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib64/python3.11/re/__init__.py", line 294, in _compile
          p = _compiler.compile(pattern, flags)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib64/python3.11/re/_compiler.py", line 743, in compile
          p = _parser.parse(p, flags)
              ^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib64/python3.11/re/_parser.py", line 980, in parse
          p = _parse_sub(source, state, flags & SRE_FLAG_VERBOSE, 0)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib64/python3.11/re/_parser.py", line 455, in _parse_sub
          itemsappend(_parse(source, state, verbose, nested + 1,
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib64/python3.11/re/_parser.py", line 841, in _parse
          raise source.error('global flags not at the start '
      re.error: global flags not at the start of the expression at position 9
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cmsis-pack-manager
Failed to build cmsis-pack-manager
ERROR: Could not build wheels for cmsis-pack-manager, which is required to install pyproject.toml-based projects
```

bp-amitchone, 2022-11-16:
I'm also experiencing this exact same issue when attempting to setup a fresh Zephyr environment on a new machine running Win 11 Pro 21H2

mbrossard, 2022-11-18:
Could you try to build the [maturin](https://github.com/pyocd/cmsis-pack-manager/tree/maturin) branch? I am not sure I have properly connected all the pieces. I need to update GitHub Actions configuration, and `setuptools_scm` needs to be replaced.

banduccm, 2022-11-19:
@mbrossard  I'll give it a shot. Initial attempts were unsuccessful, as Fedora has deprecated python2 pip, and I don't have a good way to get a python2-compatible version of setuptools on this machine. I'll try to make some time to get a VM going to try the build.

Just for the sake of completeness, I can confirm that running `setup.py` with python3 doesn't work :)

mbrossard, 2022-11-20:
> Initial attempts were unsuccessful, as Fedora has deprecated python2 pip, and I don't have a good way to get a python2-compatible version of setuptools on this machine. I'll try to make some time to [...]

`cmsis-pack-manager` does not support Python 2.x anymore and you probably need Python 3.7 with `maturin` (Python 3.6 is EOL).

To test the `maturin` you need the rust compiler installed (https://rustup.rs/) and Python with `pip`, then it should be as simple as:
```
pip install .
```

banduccm, 2022-11-20:
Oh! I'm sorry, I misunderstood and went with the readme.

I tried again with maturin, and the build appears to have worked:
```
$ pip install .
Defaulting to user installation because normal site-packages is not writeable
Processing /home/mbanducci/repos/cmsis-pack-manager
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting appdirs<2.0,>=1.4
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Requirement already satisfied: cffi in /usr/lib64/python3.11/site-packages (from cmsis-pack-manager==0.5.0) (1.15.1)
Requirement already satisfied: pyyaml<7.0,>=6.0 in /usr/lib64/python3.11/site-packages (from cmsis-pack-manager==0.5.0) (6.0)
Requirement already satisfied: pycparser in /usr/lib/python3.11/site-packages (from cffi->cmsis-pack-manager==0.5.0) (2.20)
Requirement already satisfied: ply==3.11 in /usr/lib/python3.11/site-packages (from pycparser->cffi->cmsis-pack-manager==0.5.0) (3.11)
Building wheels for collected packages: cmsis-pack-manager
  Building wheel for cmsis-pack-manager (pyproject.toml) ... done
  Created wheel for cmsis-pack-manager: filename=cmsis_pack_manager-0.5.0-py3-none-linux_x86_64.whl size=3739428 sha256=774bf0d470b7f25b2ffe540a093ee8e0f74df30fbaa08d1716f45129fded66e4
  Stored in directory: /home/mbanducci/.cache/pip/wheels/36/fc/b2/1799e56fe913dedc539a78b08828ead292fbb3ea77e774a4f5
Successfully built cmsis-pack-manager
Installing collected packages: appdirs, cmsis-pack-manager
Successfully installed appdirs-1.4.4 cmsis-pack-manager-0.5.0
```

PR #195 (2022-12-17): Replace Milksnake by Maturin
- Replaces Milksnake by Maturin to build bindings (fixes #187, fixes #188, and fixes #194).
- Pins Rust version to 1.63.0 (version 1.64.0 raised GLIBC version requirements that are not compatible with the `manylinux` version used), this should fix building wheels in CI.
- Upgrade version of `cibuildwheel`. Since `cmsis-pack-manager` does not use Python API, the bindings works independently of the version of Python. This is detected by `cibuildwheel` since version [2.8.0](https://github.com/pypa/cibuildwheel/releases/tag/v2.8.0). Upgrading the version of This will r [... cut]
closed 2022-12-17 by flit