pyocd/cmsis-pack-manager #207: 0.5.2: Unable to install wheel (cmsis_pack_manager/cmsis_pack_manager/__init__.py is not mentioned in RECORD)

dvzrv, 2023-01-30:
Hi! When trying to package 0.5.2 for Arch Linux I ran into the following problem when trying to install the built wheel:

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/usr/lib/python3.10/site-packages/installer/__main__.py", line 98, in <module>
    _main(sys.argv[1:], "python -m installer")
  File "/usr/lib/python3.10/site-packages/installer/__main__.py", line 94, in _main
    installer.install(source, destination, {})
  File "/usr/lib/python3.10/site-packages/installer/_core.py", line 96, in install
    for record_elements, stream, is_executable in source.get_contents():
  File "/usr/lib/python3.10/site-packages/installer/sources.py", line 158, in get_contents
    assert record is not None, "In {}, {} is not mentioned in RECORD".format(
AssertionError: In target/wheels/cmsis_pack_manager-0.5.2-py3-none-manylinux_2_34_x86_64.whl, cmsis_pack_manager/cmsis_pack_manager/__init__.py is not mentioned in RECORD
```

This is how I build and install:

```
maturin build --release --strip
python -m installer --destdir="$pkgdir" target/wheels/*.whl
```

dvzrv, 2023-01-30:
FWIW: This still works as intended with 0.5.1

dvzrv, 2023-04-18:
@mbrossard this is still broken unfortunately.
The resulting wheel contains duplicate dirs (however that is even possible...):

```
  Length      Date    Time    Name
---------  ---------- -----   ----
     3688  2023-04-18 21:12   cmsis_pack_manager-0.5.2.dist-info/METADATA
       98  2023-04-18 21:12   cmsis_pack_manager-0.5.2.dist-info/WHEEL
    10173  2023-04-18 21:12   cmsis_pack_manager-0.5.2.dist-info/license_files/LICENSE
     8725  2023-04-18 21:12   cmsis_pack_manager/pack_manager.py
     2267  2023-04-18 21:12   cmsis_pack_manager/cmsis_pack_manager/ffi.py
      136  2023-04-18 21:12   cmsis_pack_manager/cmsis_pack_manager/__init__.py
    13517  2023-04-18 21:12   cmsis_pack_manager/__init__.py
      136  2023-04-18 21:12   cmsis_pack_manager/cmsis_pack_manager/__init__.py
     2267  2023-04-18 21:12   cmsis_pack_manager/cmsis_pack_manager/ffi.py
142395536  2023-04-18 21:12   cmsis_pack_manager/cmsis_pack_manager/native.so
     1053  2023-04-18 21:12   cmsis_pack_manager-0.5.2.dist-info/RECORD
---------                     -------
142437596                     11 files
```

flit, 2023-04-19:
@dvzrv Sorry I didn't notice this! 🙇🏽 

0.5.2 changed the way CPM is built to move away from deprecated and unsupported milksnake. @mbrossard will have to look at this since he's the Rust dude. (Sadly, I haven't had time to learn it yet…)

dvzrv, 2024-04-06:
We are now using auto-generated source tarballs (https://rfc.archlinux.page/0020-sources-for-python-packaging/).
However, this comes with other issues (https://github.com/pyocd/cmsis-pack-manager/issues/218 and https://github.com/pyocd/cmsis-pack-manager/issues/226).
Closing, as there has been no further need for us to use sdists.

closed 2024-04-06 by dvzrv